iNavFlight/inav #5512: Hi guys,what is 0450 sencor? it showes in the telemetry.

xixijoe, 2020-03-19:
For immediate help, just ask your question on one of the following platforms:

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)

You can also read public documentations or watch video tutorials:

* [Official documentation](https://github.com/iNavFlight/inav/tree/master/docs)
* [Official Wiki](https://github.com/iNavFlight/inav/wiki)
* [Video series by Painless360](https://www.youtube.com/playlist?list=PLYsWjANuAm4qdXEGFSeUhOZ10-H8YTSnH)
* [Video series by Paweł Spychalski](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLloACrA6f1_daCjhqY2x0fB)
* [How to setup INAV on a flying wing](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLn_lCX7UB2OPLzR4w5G4el3)
* [How to setup INAV on a 5" quad](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLl9M6Vm-ZNLMCr6_9yNVHQG)

snaewe, 2020-03-19:
"Flight Path Vector" aka. "GPS heading" aka. "Course over ground"

xixijoe, 2020-03-20:
> "Flight Path Vector" aka. "GPS heading" aka. "Course over ground"

Can you tell me how to use it?

snaewe, 2020-03-20:
Well, how do you "use" other telemetry sensors?
Display it on a telemetry page for example.

xixijoe, 2020-03-20:
Most of sensors has a name with an unit that you can select. Or something like 0420, it is the distance from home. For these 2 sensors That I was asking, how can I change the name and what is the unit to select?

snaewe, 2020-03-20:
Name it whatever you want (e.g. CourseOG). It's an angle so the unit is degrees but it's actually scaled by 10 (deci-degrees).

closed 2020-03-21 by DzikuVx

xixijoe, 2020-03-22:
one more question, where I can set to show this degrees on the lua and the OSD? thank you