iNavFlight/inav #5514: Matek F405-CTR + Quad + Servo S6

DangerD1024, 2020-03-19:
Reopening issue, Servo S6 still doesn't output any PWM, tested with servo tester. Using MATEKF405_SERVOS6 firmware.
[image]

DangerD256, 2020-05-21:
A bit weird, Servo S6 is nobody using at all?

AlexeyII, 2020-05-21:
I would like to understand how to assign a servo to S6. I'm interested!

DangerD1024, 2020-07-21:
> I would like to understand how to assign a servo to S6. I'm interested!

You need to add it to servo mixer than you can assign rc channel or make automatic stabilization (for servo gimbal)
But unfortunately servo is not working on S6 this board, works fine if you will flash arducopter. (On arducopter you can even use it on S5 so you will have two servos)

craigmunday, 2020-07-29:
Hi @DangerD1024 , I also have a Matek F405 CTR flight controller and just tested the S6 pad with the MATEKF405 SERVOS6 2.5.1 target.  I can see a valid PWM signal on the oscilloscope for S6.  It seems to be working.

DangerD1024, 2020-07-30:
> Hi @DangerD1024 , I also have a Matek F405 CTR flight controller and just tested the S6 pad with the MATEKF405 SERVOS6 2.5.1 target. I can see a valid PWM signal on the oscilloscope for S6. It seems to be working.
> 
> Here is the LOW signal
> [image]
> 
> Here is the HIGH signal
> [image]

You have quadcopter in mixer or for plane?

craigmunday, 2020-07-30:
Here is my mixer configuration for my quadcopter:
[image]

closed 2020-11-07 by stale[bot]