iNavFlight/inav #5516: Instability / oscillations in all navigation modes on MR

stronnag, 2020-03-20:
inav development branch exhibits unstable oscillations in any navigation mode on MR.
In performing this test to validate #5515, #5499 and #5485, the test hex was based in current development, including #5508 (which does not appear to make any difference, the behaviour is no different from tests a week earlier). Tests from c. a month (JUMP, PH-TIMED) ago do not exhibit this undesired behaviour.

I doubt (but do not discount) that any of the JUMP, PH_TIMED or LAND changes are the cause, as other unrelated nav modes (WP -> WP, RTH, PH) are also affected.

BBL attached. The instability is also quite apparent to a ground observer (and more scary that a visual log replay suggests). See also comments in #5485
[LOG00007.TXT](https://github.com/iNavFlight/inav/files/4360422/LOG00007.TXT)

digitalentity, 2020-03-21:
Would you be able, by any chance, to bisect and find the offending commit?

stronnag, 2020-03-21:
> Would you be able, by any chance, to bisect and find the offending commit?

I'll see if I can get a quite afternoon during the week to setup in the local field with tricopter and laptop.

DzikuVx, 2020-03-21:
[LOG00007.zip](https://github.com/iNavFlight/inav/files/4363360/LOG00007.zip)

@stronnag I compiled based on 7655d1871ec16a5fe1db6caa12228afed9160520 and did some PosHold tests. Visual assesment is hard since the quad is not the one I'm doing any NAV. I'd say it looked much better than before #5508 . 
Could you take a look at the log and try to find out if it like the symptoms you observed. For me it's not

stronnag, 2020-03-21:
The oscillations I'm seeing are +/- 20°pitch +/- 10° roll at around 1 Hz, your log looks very calm by comparison.
BTW my multirotor is tricopter, I've not tried a quad.

DzikuVx, 2020-03-21:
It got dark before I was able to finish bisect. The candidates are:
f80e44ac4149f9db606b74c8091b2e155865d0fd
746d7423e8f27c012380f11eb6a8d793026edb9c

Nothing obviously wrong in either of them. Current candidate from in-head debugging is constrain put on pid integrator. But it's far fetched

DzikuVx, 2020-03-22:
I finished bisect and looks like the first bad commit is 746d7423e8f27c012380f11eb6a8d793026edb9c after all. The one where iterm limiting was introduced. Investigating deeper

stronnag, 2020-03-23:
Fixed by #5524

closed 2020-03-23 by stronnag

LELAUMAR, 2020-03-26:
I´m having the same problems as described here in #5516. Luckily you seam to have found a solution to this problem. Sadly I don't understand how to (" Remove PID iterm constrain for NAV PID controller #5524 "). Could you please help me out and explain where I need to change the code and how to apply it? 
Realy looking forward to fly my self build Quad with great Navigation.

stronnag, 2020-03-26:
If you are running a release version of inav, the problem here is not relevant, for 2.5.0 it is fixed already in the development branch.

LELAUMAR, 2020-03-26:
Thanks for this quik response!
I have a Mateksys F722 FC flashed with the 2.4(stabel) INAV version via the 2.4.1 Configurator and do have the Instability in all Navigation modes as described. So what should I do to get rid of these oscilations?
Should I flash a release candidate? Where would I get it from since the latest version shown in the configurator is the one I`m running.