iNavFlight/inav #5520: 6S battery on 4S motors - iNAV

boboscorey, 2020-03-21:
I have a 6S setup but airplane motors are usually rated at 2-4S, in betaflight there's a CLI command than can limit the motor output.

In other forums people say that I can do it by lowering the throttle input but others say this is not safe.

Please tell me if there's a way to limit the motor output or please add this feature for iNAV as well!

swissfreek, 2020-03-21:
I've been thinking about this as well. Couple thoughts:

1) Motors don't really care about voltage. They care about amps. There's not really any such thing as a "4S" motor. So those 4S airplane motors will work just fine on 6S *as long as*...

2) The amp draw must stay within limits. What this means in a practical sense is that if you have a, say, 2450kv motor that you normally run on 4S and suddenly you put 6S on it without changing the prop, you could now be over-propped and burn up the motor because the motor draws too much current.

In Betaflight it makes sense to have a motor output limit, because we are usually talking about quads. They constantly control themselves by regulating the throttle of the motors. If you use throttle limit to, say, 75%, that's on the "input" side, so it prevents *you* from sending the motors to more than 75%. But the flight controller could still tell the motors to go to 100% to correct for a wind gust or whatever, and then bad things could happen. Motor output on the other hand, is on the "output" side of the flight controller, so *nothing* can make the motors go to more than 75% throttle, or whatever you set it to.

On iNav (or more specifically when dealing with planes), the flight controller really doesn't mess with the throttle that much. So, using throttle limit is way less dangerous, because with the exception of auto modes, only you control the throttle. iNav also has settings that dictate maximum throttle in auto modes. So, long story short, I don't think it would be that dangerous to do on an airplane.

The question now is what to set max throttle to. I haven't tested that much, but if we use the same theory that folks use to get a "desired kv", if you have a motor that flies well on 4S, and you want the same output on 6S, you just need to reduce max throttle/desired kv to 66%, since 4 is 66% of 6 (and 14.8 is 66% of 22.2, we just cancel out the 3.7v/cell so you can do the math in your head). Likewise, if you have a 4S plane that you want to fly on a 5S battery, then your max throttle would be 80%.

What would be really nice is if iNav had the auto-voltage profile switching mechanism like BetaFlight has, so that you could set your plane up for 4S, then plug in a 5S or 6S battery and just fly it without having to go in and change settings between packs. But hey, we can dream...

elepete, 2020-03-21:
Well, there is one thing to have in mind. Any motor has a resistive load that is determined by the diameter of the copper wire in winding of the poles.
When you run a higher voltage the ohmic (resistive) losses go up and the power-loss or motor temperature goes up due to that.
So in general you should check the temperate of the motor after short flights, if you are not sure on the motor / voltage combination.
I am not an expert in motor / magnetic field design, but I can imagine this also can have negative effects.

DzikuVx, 2020-03-21:
you can limit throttle with `throttle_scale`. Setting it to `0.66` limits it to the desired value.
However, please remember that this is not limiting motor output or limiting voltage on the coils in any way.

closed 2020-03-21 by DzikuVx

expipiplus1, 2021-04-08:
@boboscorey Blheli_32 has a current limiting feature you may want to check out (for ESCs with current sensing). Not sure if AM32 has it...