iNavFlight/inav #5521: RC throttle command above max_check does not send max_command to the ESCs

Avegawanderer, 2020-03-21:
## Current Behavior
It seems that max_check threshold for throttle is missing or working incorrect.

## Steps to Reproduce
1. Ensure max_throttle is set to 1850 (default)
2. Ensure max_check is set to 1900 (default)
3. Arm and observe motor outputs on the outputs tab (no battery required)
4. Move throttle stick slowly. Notice that when RC command exceeds max_check, ESC command does not stick to max_throttle. It continues increasing, reaching 100% (1850) when RC command equals 2000

## Expected behavior
After RC command exceeds max_check, constant value of max_throttle should be sent to ESCs

## Suggested solution(s)
Review and fix the code for ESC control logic

## Additional context
<script src="https://pastebin.com/embed_js/bHdLzvJd"></script>

---

Entering CLI Mode, type 'exit' to return, or 'help'

# version
# INAV/MATEKF722SE 2.4.0 Feb 11 2020 / 22:53:38 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

DzikuVx, 2020-03-22:
yes, this is correct behavior. The is no max_check on throttle when motor output is concerned

closed 2020-03-22 by DzikuVx

Avegawanderer, 2020-03-22:
Is this really correct behavior? It means 100% throttle is achieved for maximum value specified by rxrange (2000 by default). In a perfect world that would be OK but in the real there are variuos drifts which may lead to reduce of maximum signal output by TX to some lower value. I also suggest that many pilots would prefer a little stick range for maximum throttle on the top of the stick throw (as well as for zero throttle). Along with recomendation to set transmitter endpoint exactly to 1000 - 2000, this may result in 99-100% ESC wobbling when stick is in the maximum throttle position. This probably is not true for analog ESC protocols as many (if not all) ESCs add some margin for PWM range obtained from calibration.
Yes, this may be bypassed by setting throttle channel rxrange to a bit narrower, with maximum of ~1950 (or increasing maximum on the ransmitter), but that is generally wrong.
The documentation explicitly states:
> max_check - Throttle positions above this level will send max_command to the ESCs.
I think the approach to the maximum throttle value should be similar to minimum throttle. Throttle range should be started with idle_throttle at min_check, increasing to max_throttle at max_check. Below min_check ESCs get min_command (MOTOR_STOP) or idle_throttle (MOTOR_STOP disabled), Above max_check ESCs get max_throttle.
By the way, maybe all throttle settings should be specified in %, not us units. That would make setup easier and cleaner.