iNavFlight/inav #5526: Servo output not logged in blackbox

pulquero, 2020-03-22:
## Current Behavior
Servo output not present in blackbox log.

## Steps to Reproduce
1. Enable blackbox logging when inflight. 
2. Download log afterwards 
3. Notice servo output is not recorded 

## Expected behavior
Blackbox log contains servo output. 

## Additional context

---

- FC Board name and vendor: F405SE
- INAV version string:

stronnag, 2020-03-22:
It is, since inav 2.2

pulquero, 2020-03-22:
Example log: [LOG00011.TXT](https://github.com/iNavFlight/inav/files/4366485/LOG00011.TXT)

stronnag, 2020-03-23:
Looks like servos to me, all 16:
```
$ bbsummary -d ~/dl/LOG00011.TXT 
Log      : LOG00011.TXT / 1
Craft    : "AR900" on 2020-02-23T17:00:02.822+00:00
Firmware : INAV 2.4.0 (bcd4caef9) MATEKF405SE of Feb 11 2020 22:58:10

 "loopIteration"
 "time (us)"
 "axisRate[0]"
 "axisRate[1]"
 "axisRate[2]"
 "axisP[0]"
 "axisP[1]"
 "axisP[2]"
 "axisI[0]"
 "axisI[1]"
 "axisI[2]"
 "axisD[0]"
 "axisD[1]"
 "axisD[2]"
 "fwAltP"
 "fwAltI"
 "fwAltD"
 "fwAltOut"
 "fwPosP"
 "fwPosI"
 "fwPosD"
 "fwPosOut"
 "rcData[0]"
 "rcData[1]"
 "rcData[2]"
 "rcData[3]"
 "rcCommand[0]"
 "rcCommand[1]"
 "rcCommand[2]"
 "rcCommand[3]"
 "vbat (V)"
 "amperage (A)"
 "BaroAlt (cm)"
 "rssi"
 "gyroADC[0]"
 "gyroADC[1]"
 "gyroADC[2]"
 "accSmooth[0]"
 "accSmooth[1]"
 "accSmooth[2]"
 "attitude[0]"
 "attitude[1]"
 "attitude[2]"
 "motor[0]"
 "servo[0]"
 "servo[1]"
 "servo[2]"
 "servo[3]"
 "servo[4]"
 "servo[5]"
 "servo[6]"
 "servo[7]"
 "servo[8]"
 "servo[9]"
 "servo[10]"
 "servo[11]"
 "servo[12]"
 "servo[13]"
 "servo[14]"
 "servo[15]"
 "navState"
 "navFlags"
 "navEPH"
 "navEPV"
 "navPos[0]"
 "navPos[1]"
 "navPos[2]"
 "navVel[0]"
 "navVel[1]"
 "navVel[2]"
 "navAcc[0]"
 "navAcc[1]"
 "navAcc[2]"
 "navTgtVel[0]"
 "navTgtVel[1]"
 "navTgtVel[2]"
 "navTgtPos[0]"
 "navTgtPos[1]"
 "navTgtPos[2]"
 "navSurf[0]"
 "energyCumulative (mAh)"
 "flightModeFlags (flags)"
 "stateFlags (flags)"
 "failsafePhase (flags)"
 "rxSignalReceived"
 "rxFlightChannelsValid"
 "hwHealthStatus"
 "powerSupplyImpedance"
 "sagCompensatedVBat"
 "wind[0]"
 "wind[1]"
 "wind[2]"
 "IMUTemperature"
 "baroTemperature"
 "sens0Temp"
 "sens1Temp"
 "sens2Temp"
 "sens3Temp"
 "sens4Temp"
 "sens5Temp"
 "sens6Temp"
 "sens7Temp"
 "GPS_fixType"
 "GPS_numSat"
 "GPS_coord[0]"
 "GPS_coord[1]"
 "GPS_altitude"
 "GPS_speed (m/s)"
 "GPS_ground_course"
 "GPS_hdop"
 "GPS_eph"
 "GPS_epv"
 "GPS_velned[0]"
 "GPS_velned[1]"
 "GPS_velned[2]"

Disarm   : SWITCH

```

DzikuVx, 2020-03-23:
[image]

Just like @stronnag said: all 16 servo outputs  logged.

I think we can close it

closed 2020-03-23 by DzikuVx

pulquero, 2020-03-23:
Thanks, that helps. I guess I focused on servo 0 & 1 rather than checking the others closely enough.