iNavFlight/inav #5529: request for keep the waypoints instead of getting it from eeprom everytime new fly.

xixijoe, 2020-03-24:
## Current Behavior

## Desired Behavior

## Suggested Solution

## Who does this impact? Who is this for?

## Additional context

DzikuVx, 2020-03-24:
1 - please try to create tickets with title and body, not only with title
2 - waypoints altready can be stored in EEPROM

closed 2020-03-24 by DzikuVx

xixijoe, 2020-03-24:
1,Okay body and title.
2,Yes, eeprom can be stored. The problem is you need to get it from eeprom everytime new fly.This is not a convenient.

stronnag, 2020-03-24:
> 2,Yes, eeprom can be stored. The problem is you need to get it from eeprom everytime new fly.This is not a convenient.

Moving one stick to a specific position for c. 1 second is hardly inconvenient.

xixijoe, 2020-03-24:
yes thank you....

xixijoe, 2020-03-25:
stick command not funtional when arm switch is assigned. and arm swith is recomanded.......not convenient.

stronnag, 2020-03-25:
You cannot load WPs when armed: period.
When unarmed, the stick loading works. with arming assigned to a switch.

xixijoe, 2020-03-25:
I tried. it looks works. I will bring it on the field next time flying. Thank you for your sharing.

xixijoe, 2020-03-27:
It works, thank you