iNavFlight/inav #5531: KAKUTE F7 Servo Output

geniusboywonder2020, 2020-03-24:
Hi Guys, i am trying to control a gimbal from my kakute F7. I have a storm32 gimbal controller and with the current wiring (rc pins to motor 5 and 6) i can control the gimbal in betaflight. I cannot control the gimbal in inav. I have my transmitter set to have 2 controls and i can successfully change the values in the servos tab. I cannot however get any movement on the gimbal. In the servos tab is shows output as S5 and S6 which is what i am using.

Where can link s5 and s6 to the motor pins.

I have adapted the firmware files to assign the pins but they still show as free in resources. They are both on different timers as well.

DzikuVx, 2020-04-14:
Please provide diff and screenshot of Mixer tab. Also, have you see this: https://youtu.be/VmbyfV8UyfQ

geniusboywonder2020, 2020-04-14:
Hi Pawel, i have watched that video several times and i cannot get it working.

I have attached the diff and the mixer tab screenshot as requested.
`diff

# version
# INAV/KAKUTEF7 2.4.0 Mar 24 2020 / 15:56:40 (e31c9f2)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# start the command batch
batch start

# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix 0 0 15 100 0 -1
smix 1 2 16 100 0 -1

# servo

# logic

# gf

# feature
feature -AIRMODE
feature PWM_SERVO_DRIVER
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100
aux 1 1 1 900 1725
aux 2 4 1 1300 1700
aux 3 7 1 900 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 20 0 V
osd_layout 0 2 0 0 V
osd_layout 0 7 13 13 V
osd_layout 0 13 24 0 V
osd_layout 0 14 0 13 V
osd_layout 0 20 18 14 V
osd_layout 0 21 1 14 V
osd_layout 0 22 13 2 V
osd_layout 0 23 16 2 V
osd_layout 0 28 23 13 V
osd_layout 0 30 2 15 V

# master
set gyro_hardware_lpf = 188HZ
set acc_hardware = MPU6000
set acczero_x = 20
set acczero_y = 14
set acczero_z = 4
set accgain_x = 4169
set accgain_y = 4161
set accgain_z = 4115
set mag_hardware = QMC5883
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 16
set serialrx_provider = CRSF
set motor_pwm_rate = 16000
set motor_pwm_protocol = DSHOT600
set model_preview_type = 3
set servo_pwm_rate = 330
set applied_defaults = 1
set gps_sbas_mode = AUTO
set vtx_band = 1
set vtx_power = 5
set vtx_freq = 5865

# profile
profile 1

set heading_hold_rate_limit = 20

# battery_profile
battery_profile 1

set battery_capacity = 2600

# end the command batch
batch end

# `
[image]

closed 2020-06-28 by stale[bot]