iNavFlight/inav #5549: Bidirectional DShot

HeikoBischoff, 2020-03-29:
## Current Behavior
To use RPM filter, the ESC telemetry must be connected to the flight controller.

## Desired Behavior
Most ESCs with BLHeli_32 and BLHeli_S are supporting bidirectional DShot. The ESC telemetry will be delivered to the flight controller by the motor wire. 

## Suggested Solution
Betaflight is supporting bidirectional DShot since version 4.0/4.1. 
It would be great to implement/adapt this feature in iNav.

## Who does this impact? Who is this for?
Mostly all multirotor builds which want to take advantage of RPM filter without wiring an additonal line from the ESC to the flight controller.

## Additional context
Link to the Betaflight RPM Filter page:
https://github.com/betaflight/betaflight/wiki/Bidirectional-DSHOT-and-RPM-Filter

denisiussion, 2020-11-28:
Any chance to have it on nearest release?

harry1453, 2021-04-23:
I'm not able to work on this right now, but unless anyone else is working on this, I'm planning to tackle this next.

DzikuVx, 2021-04-23:
Be our guest mate!

padcom, 2021-07-28:
Any plans (near or far) to implement it?

harry1453, 2021-07-30:
@padcom I plan to do it, but it's quite complex and I don't understand the STM32's DMA architecture well enough at the moment. I've written an implementation for ESP32 so that I understand what goes on on the wire, I'm just fighting with DMA. Consider it a far plan, not a near one.

AADDZZ, 2021-11-21:
@harry1453 Hi bro ! Any News about this feature ? Are you still working on it ? 

Lot of people like me are waiting this to switch from Betaflight to Inav.

RipperGOT, 2022-01-27:
Please, implement this option, it would be very useful.  Even for esc like the Hobbywing rotor 60a blheli_32 which has no telemetry output. 

 Thanks guys!

krizw, 2022-05-12:
Any news on this feature being added?

r1000ru, 2023-04-03:
Since now the blheli s esc with bluejay firmware and the blheli32 esc have begun to give not only RPM, but also temperature via bidirection dshot, performing this issue would be very useful.

Alex-TV, 2023-06-07:
There is a clone of the repository where this has already been done. Any chance to include it here?

lida2003, 2023-06-21:
2023, still don't have bdshot supported yet?

> There is a clone of the repository where this has already been done. Any chance to include it here?

fritzrichter, 2023-08-24:
Hey guys, it's my first comment here. I just switched from Betaflight to iNav as especially as a beginner I love to have the position fix feature to pause or even land easily, quite nice and Betaflight guys seem not to want to invest into this as they "mainly focus on performance" - oh man. Anyway, the bdshot is something which seems you want to have and I was quite sad that this is not supported yet, quite old request though. Any update or plans? Quick question, as Betaflight has integrated it and it's open source as well, isn't it easy to get the underlaying idea from their source code? Why is it so hard to get the best out of these two worlds and let each other inspire a bit?

danarrib, 2023-08-24:
> Hey guys, (...) I just switched from Betaflight to iNav (...) Quick question, as Betaflight has integrated it and it's open source as well, isn't it easy to get the underlaying idea from their source code? 

Check this out: https://github.com/iNavFlight/inav/wiki/INAV-for-BetaFlight-users

closed 2024-02-02 by DzikuVx

marianomd, 2024-02-02:
This is good news @DzikuVx ! What is the PR implementing this? Thanks

r1000ru, 2024-02-04:
It is issue is really completed? May be remove support blheli32 telemetry and add bdshot, that use and blheli32, and bluedjay, have rpm, voltage, currency and temperature?

MrD-RC, 2024-02-04:
Why remove the telemetry option? That works fine. It would make no sense to remove it. 

@DzikuVx has said a few times in his live streams that there is no intent to implement bidirectional DSHOT. I’m sure he mentioned that is can cause issues with the communications. Especially if longer cable runs are used.

course-up, 2024-02-05:
I am little confused, the feature request is closed without comment. Is that good or bad, does that mean that it will be implemented or that the feature request will be just ignored?

lida2003, 2024-02-05:
> I am little confused, the feature request is closed without comment. Is that good or bad, does that mean that it will be implemented or that the feature request will be just ignored?

`DShot telemetry is supported, but not Bi-directional single-wire telemetry.` from https://github.com/iNavFlight/inav/wiki/INAV-for-BetaFlight-users

@DzikuVx @MrD-RC @course-up @r1000ru I think it's closed as inav'll NOT support this feature, PLEASE correct me, if I'm wrong. 

a) As I didn't see any PR about bdshort supported right now.
b) @MrD-RC has stated "might cause issues with the communications, especially if longer cable runs are used."

marianomd, 2024-02-10:
So when closing, @DzikuVx should add the correct label, in this case **wontfix** or similar.

Patronics, 2024-09-12:
> There is a clone of the repository where this has already been done. Any chance to include it here?

Can you send a link to this fork? Has it been updated to maintain compatibility with the latest INAV release? 

Also can someone (maybe @MrD-RC or @DzikuVx) clarify why apparently bidirectional dshot is so problematic that it can't even be added as an option? 

My ESC doesn't support the older telemetry style, so if I switch from Betaflight to INAV (which I really want to do, the design goals are much closer to fitting my octocopter's use case), I'll have no telemetry available at all. Won't that cause bigger issues than just a vague "might cause issues with communications on long cable runs". Betaflight doesn't seem to have had issues with bdshot, so why would the issues only be caused in INAV?

course-up, 2024-10-08:
@DzikuVx has a peculiar opinion on some features. One of these features is **_Bidirectional Dshot_** and he thinks that it is not needed and blocks intentions to improve iNav.
A great video (by Mark Spatz) covers the benefits of Bidirectional DShot: https://youtu.be/eoxceps6Dhk?t=194

sensei-hacker, 2024-10-08:
@course-up  maybe you didn't see what DzikuvX said right here, in this issue?

I'll quote it for you. HarryJPH said he planned to work on and DzikuvX said:
Be out guest mate!

https://github.com/iNavFlight/inav/issues/5549#issuecomment-825857366

That's right here in this page!

There are some challenges, particularly when people run DSHOT600, which updates each motor 36,000 times per second. (Much faster than the loop time, and therefore mostly redundant, but the same people who want bidirectional DShot tend to do that.) Additional noise on the wire doesn't help when you're already running the communication at a very high speed.

But what he has said is you're more than welcome to work on it. He's not blocking anything. He isn't going to spend his time doing it. I'm not going to spend my time doing it. He has invited you to do it if you want to. Just in case you're confused, I don't have any obligation to spend my weekends working on this for you. You can do it if you want to. If you don't want to, please stop making up BS. Thanks.

Patronics, 2024-10-08:
Well, it would be a bit more likely that someone would do it if this issue wasn't "closed as completed", making it harder for anyone wanting to contribute to find it and realize that it would be a useful feature that hasn't already been implemented! 

Not everyone that wants to get telemetry from their motors (with the only protocol their ESC supports, despite being seemingly the most widely recommended 4-in-1 ESC that I've seen) is the sort of person who also wants the fastest and potentially noisy communication protocol, despite your overgeneralization. I think that even DSHOT150 with telemetry would potentially be more important than DSHOT600 without it, would it not? 

In the interest of providing a reference to make it easier for anyone who's able to contribute, here's [some relevant commits](https://github.com/search?q=repo%3Abetaflight%2Fbetaflight++dshot+Telemetry+sort%3Aauthor-date+&type=commits&s=committer-date&o=asc) from the betaflight git repo (start at the 3rd result, the initial commit adding bidirectional Dshot to Betaflight). Some additional results can be found by searching "bidirectional dshot", but it seems that the most substantial commits I found simply referred to the feature as dshot telemetry. 

There's also [this repo](https://github.com/symonb/Bidirectional-DSHOT-and-RPM-Filter) and associated [blog post](https://symonb.github.io/docs/drone/ESC/ESC_prot_impl_2_2) that extensively documents a reference implementation of BDShot, and could be a valuable resource. 

Hopefully this is helpful for someone.

course-up, 2024-10-08:
Bidirectional DShot has been around for quite a long time and many ESCs do not support ESC telemetry via UART. Therefore, the feature request of iNav users is understandable.

Pawel's opinion on bidirectional DShot and RPM filters is well known from live streams and comments. Closing the ticket without any comment also speaks for itself. I don't expect the feature to come to iNav, at least not in the near future. But Betaflight is catching up fast: Alt Hold is in the works and Mark Spatz was mentioning that Pos Hold is also on the roadmyp. For me, that would be perfectly fine.

rts18, 2024-10-10:
Guys! If you follow this project you should know INAV has a much smaller development team than Betaflight, with most flying fixedwings.
Unfortunately the performance benefits of BDshot are not realized by much of the team. And even if they were, That would still be no guarantee it could be added due to each developer having a specific coding skills set. There is no shame in saying each to his own ability.
INAV has only one developer continually adding to the navigation code. With his efforts sustaining the whole project in my opinion. He is little known due to his lack of presence on social media. So don't disrespect his efforts by forcing a guilt trip and complaining in ignorance.

**If you know any developers who have a coding knowledge of DMA architecture; ask them if they'd be willing to contribute to the INAV project.** 

>Alt Hold is in the works and Mark Spatz was mentioning that Pos Hold is also on the roadmyp. For me, that would be perfectly fine.

@course-up Since you follow Mark, Take note of his words of wisdom in this [video](https://www.youtube.com/watch?v=xtAYv8myIZA)

IgnacioHR, 2025-11-02:
[image]

Adding a field‑test report that may help reproduce and prioritise this.

Use case: 5" long‑range quad where iNAV’s RTH/navigation stack is the main reason to use iNAV over BF for LR safety.

Hardware:
• FC: SEQURE H7 V2 (STM32H743)
• ESC: AM32 4‑in‑1 (supports analog PWM and DShot300/600)
• Motors/props: typical 5" setup

Observed on iNAV 8.0.1 (H743 target):
• In ACRO, with steady zero setpoint and no status changes, I get periodic paired‑motor kicks (motors 1+3 vs. 2+4). Total thrust remains roughly constant; the gyro then shows the corrective reaction as the controller brings the craft back.
• The effect repeats every few seconds and is independent of RC input.
• The same airframe + ESC works perfectly on Betaflight with DShot600 + bidirectional telemetry (RPM filter enabled), so this looks like an iNAV H743 output/timing issue rather than hardware.

What I’ve tried so far:
• DShot150/300/600; enabling/disabling RPM filter; adjusting loop timing; lowering D‑term LPF (e.g., 80 Hz). Symptoms do not disappear.
• “Standard” PWM is unreliable on this ESC (startup and idle jitter), so I’m not flying PWM.

Target‑specific details that may matter:
• On SEQURE H7 V2, all four motors are mapped to TIM3 with DMA; other pads on this target lack DMA (per target sources). That might explain sensitivity to jitter on H743 at higher DShot rates.

Requests:
1) Please consider adding native Bidirectional DShot to iNAV. It would remove the UART dependency for RPM telemetry, align with modern ESCs (AM32/BLHeli_32), and improve robustness of the RPM filter pipeline.
2) I’m happy to provide short Blackbox logs and `tasks` output (and to run A/B tests across minor versions) to help pinpoint this on H743.

Thanks for considering this; I’m available to test patches and report back quickly.