iNavFlight/inav #5554: Instability in all navigation modes, fix?

LELAUMAR, 2020-03-30:
I´m having the same problems as described in #5516. Luckily you seam to have found a solution to this problem. Sadly I don't understand how to (" Remove PID iterm constrain for NAV PID controller #5524 "). Could you please help me out and explain where I need to change the code and how to apply it.  I have a Mateksys F722 FC flashed with the 2.4(stabel) INAV version via the 2.4.1 Configurator and do have the Instability in all Navigation modes as described. So what should I do to get rid of these oscilations?  Should I flash a release candidate? Where would I get it from since the latest version shown in the configurator is the one I`m running.

_Originally posted by @LELAUMAR in https://github.com/iNavFlight/inav/issues/5516#issuecomment-604436678_

stronnag, 2020-03-30:
Same answer as #5516, which is not relevant to stable branches.

closed 2020-03-30 by stronnag

DzikuVx, 2020-03-30:
Just like @stronnag wrote: the problem #5516 was never released and was never a part of official release and does not apply here