iNavFlight/inav #5561: MOAR SPEED

DBetic-Pilot, 2020-03-31:
## Current Behavior
Currently using a 6" Multirotor (Tyro 119) with INAV and 3dr telemetry setup. When attempting to fly waypoints, the nav_auto_speed maximum is set to 20m/s, and any waypoint entered at a faster speed will drop down to 20m/s. Even in this modest configuration, the aircraft is easily capable of 30m/s, but is being firmware-limited to a max of 20. 

## Steps to Reproduce
1. set nav_auto_speed to anything greater than 20m/s

## Expected behavior
I expected the ability to drive the aircraft to the point at which a nav speed would prevent it from maintaining altitude or destabilizing. 

## Suggested solution(s)
Removal of the firmware speed limit so that users can adjust nav_auto_speed to match vehicle performance and capabilties

## Additional context
- FC Board name and vendor: Eachine Tyro 119 - MATEX F405 STD Firmware
- INAV version string:2.4.0
[image]
[image]

giacomo892, 2020-04-09:
Hi @DBetic-Pilot 

I don't think you have understood how open source projects work.

First of all the issue must have a serious title, which has not in this case.
Don't expect the developers to be your personal slaves, "bump" is really rude and increases the chances the issue will get never addressed.

The amount time we put in the project is mainly driven by the interested in the hobby in improving/implementing what we feel to address and of course since our days are made up of 24h like yours we have to take time off something else.

Since INAV is an open source project you have at least two possibilities:

1. Fix the issue by yourself and render the fix public
2. Open an issue as you did, wait patently for someone to evaluate it and perhaps fixing it.

Hoping the situation is clearer now

DBetic-Pilot, 2020-04-09:
Completely understand on both accounts. I never understood a bump to be
rude, but rather a means of simply saying that it is still if interest to
me, in case it was not seen previously. If it is taken as such by you and
the other devs, my sincerest apologies.
I will adjust the title as well if you feel the name is inappropriate.
Thanks For the honest feedback
Chris

DBetic-Pilot, 2020-04-09:
It would appear I can’t do this as my issues have been deleted... not sure
that is the most professional  response either though if you are looking to
grow a collaborative network.
Chris

DBetic-Pilot, 2020-04-11:
So I made a few minor adjustments in settings.yaml, but have run into numerous issues trying to compile (admittedly my first time compiling). I have worked through all of the issues to get MINGW32 to compile, but it fails after "Linking REVO."

When just typing "Make" i get the attached error after several minutes of compiling. My google and wiki-fu is not advanced enough to overcome this one...
[image]

Thanks
Chris

DBetic-Pilot, 2020-04-14:
I ended up moving my entire machine over to linux mint, which has simultaneous solved and created problems. I was able to compile a custom firmware, but it does not appear that the src/main/fc/settings.yaml is the only location that is influencing the nav_auto_speed parameter (line 1426). Is there anywhere else I should be looking, or can the INAV configurator over-write this once flashed? 

In the meantime, I did put together a short bash script that should help linux users automagically configure USB port serial and run INAV from the desktop. Nothing fancy, but would have saved me at least a few hours of googling when new to all this. if this is of any interest, i'd be happy to contribute it to the Wiki

DBetic-Pilot, 2020-05-10:
creating a custom firmware and modifying the nav_auto_speed parameter allowed me to take the vehicle up to 74 mph. This was on 4S, so the ability to maintain level flight was compromised, as well as some other issues like wind resistance starting to affect baro inputs to the altitude controllers.

closed 2020-05-10 by DBetic-Pilot