iNavFlight/inav #5563: Omnibus F4 V6 S1 not working

michalelektryk, 2020-04-03:
So I'm trying to set up my omnibus f4 as FC for my quad, but one output channel is not working. It's not ESC problem as I tried swapping cables around and using servos to probe. Motor outputs S3,5 and 6 work, but S1 doesn't. I tried adding more motors in mixer, but I doesn't allow to use S2 and 4 (why) and S7 and S8 seem to not be available on my board.

DzikuVx, 2020-04-14:
Please provide diff. We are not prophets you know

michalelektryk, 2020-04-15:
It literally 5" quad preset loaded with 5" mixer, but here it comes:

# diff

# version
# INAV/OMNIBUSF4V3 2.4.0 Feb 11 2020 / 22:23:56 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# start the command batch
batch start

# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix 0 0 29 30 0 -1

# servo

# logic

# gf

# feature
feature DYNAMIC_FILTERS
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 5 2 115200 9600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout

# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 90
set gyro_lpf_type = PT1
set gyro_notch1_hz = 400
set gyro_notch1_cutoff = 300
set gyro_notch2_hz = 200
set gyro_notch2_cutoff = 100
set acc_hardware = MPU6000
set acczero_x = 75
set acczero_y = -23
set acczero_z = -314
set accgain_x = 4100
set accgain_y = 4119
set accgain_z = 4025
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set blackbox_rate_denom = 32
set motor_pwm_rate = 2000
set motor_pwm_protocol = ONESHOT125
set model_preview_type = 3
set applied_defaults = 2
set gps_provider = NMEA
set mc_airmode_type = THROTTLE_THRESHOLD

# profile
profile 1

set mc_p_pitch = 48
set mc_i_pitch = 50
set mc_d_pitch = 22
set mc_p_roll = 36
set mc_i_roll = 40
set mc_d_roll = 20
set mc_p_yaw = 70
set dterm_lpf_hz = 90
set use_dterm_fir_filter = OFF
set dterm_notch_hz = 260
set dterm_notch_cutoff = 160
set d_boost_factor =  1.500
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set rc_yaw_expo = 70
set roll_rate = 75
set pitch_rate = 75
set yaw_rate = 75

# battery_profile
battery_profile 1

# end the command batch
batch end

closed 2020-06-28 by stale[bot]

MorrisR2, 2021-11-21:
Try using the OMNIBUSF4V6 target rather than OMNIBUSF4V3.