iNavFlight/inav #5566: Matek F765 WING iNAV Configurator:  Save And Reboot = no reconnection error

kniuk, 2020-04-05:
iNAV Configurator 2.4.1

2020-04-05 @ 12:45:21 -- Serial port successfully opened with ID: 3
2020-04-05 @ 12:45:21 -- MultiWii API version received - 2.4.0
2020-04-05 @ 12:45:21 -- Flight controller info, identifier: INAV, version: 2.4.0
2020-04-05 @ 12:45:21 -- Running firmware released on: Feb 11 2020 23:09:10
2020-04-05 @ 12:45:21 -- Board: M765, version: 0
2020-04-05 @ 12:45:21 -- Unique device ID received - 0x2a00353038510139353336
2020-04-05 @ 12:45:28 -- EEPROM saved
2020-04-05 @ 12:45:28 -- Device - Rebooting
2020-04-05 @ 12:45:28 -- Serial port successfully closed
2020-04-05 @ 12:45:35 -- Failed to open serial port
2020-04-05 @ 12:45:36 -- You need to upgrade your firmware before you can use the Mixer tab.

Same for different tabs, e.g. Configuration tab
Unpluging USB plug from PC and pluging it again fixes this. No iNAV configurator restarting neccessary.
Restarting iNAV Configurator without unlugging USB works as well.

Pairan, 2020-04-06:
Got something similar on MacOs while trying to flash with Configurator 2.4.1 and a version prior :

- selecting the FC
- selecting the Version
- Load firmware online
- "cant connect to Bootloader" (if I remember right!)

Unplugging the USB cable and replugging it helps, as the above procedure then works!

DzikuVx, 2020-04-14:
This will be fixed with Configurator 2.5

closed 2020-04-14 by DzikuVx