iNavFlight/inav #5572: MS5611 external baro does not work in iNav...

quetiano, 2020-04-06:
Good day to all
I have the mamba f405 mk2 stack. I have done some configuration in betaflight and finally got the gps with built in compass and baro, MS5611 to work, however, in inav, it refuses to work. I have tried resistors and changing to other baro hardware in inav and even set the clock speeds of the i2c port to various speeds and it just doesnt work. The baro is lit in red but the gps and compass is lit in blue together with accel and gyro.
Any work arounds for this !

robertha64, 2020-04-15:
You need to provide more details about your config - the inav version string from CLI at least. (Go to CLI and type version - then paste result here).

quetiano, 2020-04-15:
Thank you so much for your reply @robertha64 
Unfortunately, I damaged my mamba f405 mk2 somehow, through soldering or something else with the soldering. 

Currently our country is on lock down due to the corona virus and so I cannot order another flight controller until the lock down is over. I am testing an old apm 2.8 to see if I can perhaps use that to survive for now.

Somebody helped me regarding this problem by modding the inav code and enabled the I2C_2 ports via the uart 3 for my mamba, I had the baro SDA & SCL connected to my uart 3 rx and tx, and after flashing the modded firmware to my mamba, the baro began to work but he forgot to activate the I2C_1 which had normal SCL & SDA psychical soldering pads on the mamba. When I tried to get ahold of him, he did not reply , so I left it for the time being and decided to resolder some wire that came loose and I think thats where I caused the damage, perhaps excessive heat on one of the pads while burning the flux, or melted solder got onto one of the chips, which I cant really see anywhere.

Anyway, I learnt my lesson as I go along and now I plan to purchase a flight controller with a built in barometer so I wont have to battle like this again with this issue that I had with trying to get sensors to work on this mamba f405 mk2.

closed 2020-06-21 by quetiano