iNavFlight/inav #5576: Insane climb rate in RTH - Mode - up and down

Burnst81, 2020-04-07:
Hello Community! I´ve tried to figure it out by myself but my knowledge ends here...

I f i activate the RTH-Mode on my 3 inch quad it turns to the home direction and starts climbing like Hell. I set the RTH - Altidude to 3500 so its 35 meters. The quad begins to climb with full throttle up to 50 meters.. then begins to lose heigh to 35 meters again.. if it reaches the 35 meters it gives full throttle again till it reaches 71 meters.. then begin to descending again.. 

.. this it does again and again… whats wrong with this guy? I tried to Change climb rates etc.. but Nothing helps..

Here is my Diff / all

# diff /all

# version
# INAV/MATEKF722MINI 2.2.1 Jul  3 2019 / 22:46:46 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# logic

# feature
feature THR_VBAT_COMP
feature BAT_PROF_AUTOSWITCH
feature PWM_OUTPUT_ENABLE

# beeper
beeper -HW_FAILURE
beeper -RX_LOST
beeper -BAT_CRIT_LOW
beeper -ON_USB

# map

# serial
serial 0 2 115200 57600 0 115200
serial 1 68 115200 115200 0 115200
serial 2 2048 115200 115200 0 115200
serial 3 1024 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100
aux 1 1 2 1300 1700
aux 2 3 3 1300 1700
aux 3 9 3 1700 2100
aux 4 8 1 1700 2100
aux 5 21 5 900 1100
aux 6 11 2 1700 2100
aux 7 30 4 900 1100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 13 13 V
osd_layout 0 1 12 0 H
osd_layout 0 2 0 0 V
osd_layout 0 4 8 6 H
osd_layout 0 7 23 14 V
osd_layout 0 9 11 14 V
osd_layout 0 11 2 3 H
osd_layout 0 12 1 4 H
osd_layout 0 14 1 1 V
osd_layout 0 15 24 3 V
osd_layout 0 22 11 1 V
osd_layout 0 23 13 1 V
osd_layout 0 27 23 4 H
osd_layout 0 28 23 13 V
osd_layout 0 30 1 10 V
osd_layout 0 32 1 14 H
osd_layout 0 34 10 0 V
osd_layout 0 38 2 13 V
osd_layout 0 54 1 14 V
osd_layout 0 85 24 4 V
osd_layout 0 105 28 14 V

# master
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 100
set acc_hardware = MPU6000
set acczero_x = 48
set acczero_y = -20
set acczero_z = -187
set accgain_x = 4084
set accgain_y = 4076
set accgain_z = 3995
set align_mag = CW270FLIP
set mag_hardware = QMC5883
set magzero_x = 34
set magzero_y = 276
set magzero_z = 217
set baro_hardware = MS5611
set pitot_hardware = NONE
set rssi_channel = 16
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set min_throttle = 1070
set max_throttle = 2000
set motor_pwm_rate = 16000
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = RTH
set model_preview_type = 3
set 3d_deadband_low = 1425
set 3d_neutral = 1475
set servo_pwm_rate = 330
set disarm_kill_switch = OFF
set gps_sbas_mode = AUTO
set nav_extra_arming_safety = OFF
set nav_auto_speed = 600
set nav_auto_climb_rate = 300
set nav_rth_alt_mode = FIXED
set nav_rth_altitude = 3500
set nav_mc_hover_thr = 1250
set osd_video_system = PAL
set osd_crosshairs_style = TYPE3
set vtx_band = 5
set vtx_channel = 6
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5843

# profile
profile 1

set mc_p_pitch = 50
set mc_d_pitch = 25
set mc_p_roll = 50
set mc_d_roll = 25
set mc_p_yaw = 70
set mc_d_yaw = 20
set dterm_lpf_hz = 100
set roll_rate = 75
set pitch_rate = 75
set yaw_rate = 75
set manual_rc_expo = 100

# battery_profile
battery_profile 1

DzikuVx, 2020-04-14:
impossible to say anything without a blackbox log

britmick, 2020-04-25:
H there did  you get any where with the issue, just come back form my first test flight wit INAV all working except RTH. As soon as I  engage RTH the quad does a full punch out.  I didn't leave it to see if it cycled as the quad was getting quite high I returned to angel mode and flew it back.  I've setup Arducopter on the same quad with same hardware and all worked as it should. Omnibus F4 v6, INAV latest version setup and calibrated, modes switches set and hover throttle not touched anything else.

digitalentity, 2020-04-25:
Blackbox log is needed to debug this kind of issue.

closed 2020-07-11 by stale[bot]