iNavFlight/inav #5579: RTH For gliders

BarryCole, 2020-04-08:
## Current Behavior

## Desired Behavior

## Suggested Solution

## Who does this impact? Who is this for?

## Additional context

BarryCole, 2020-04-08:
Sorry if this is in the wrong place, I am new to this.
Would it be possible to have a RTH function that would work with a glider.
It would need ASI to monitor air speed, and use the elevator rather than the throttle to control the air speed.

Is this possible.

BC

dybiusz, 2020-04-08:
Hi Barry, you can try to set altitude for RTH as AT_LEAST_LINEAR_DESCENT =2000 (or something else) and good gliding slope (this in "Fixed Wing Navigation Settings" : Max. dive angle [degrees]) and if you are above this 2000cm in decent whether condition you have chance to see your glider again.
If your magnetometer is off (seems popular for FW.), you need to gain some horizontal speed concerning ground,  in order to let your gps to calculate course and make some maneuvers. By strong wind it could not work , like didn't work on my Z84 with small throttle set because motor did not compensate wind and my F4ominbus went crazy. 
Do some tests- it might work.

BarryCole, 2020-04-10:
I want to put this in a 5Mtr glider for aerotowing, in case I lose sight of it at 1000 ft+. I think it will need the ASI to ensure that it does not over speed, or stall.

dybiusz, 2020-04-14:
In your case I would install R9 STAB OTA + gps + ASI , then add to transmitter some lua script (presume you have tarranis ) to display direction , home direction , ground speed ,air speed ,home distance and the altitude at the same telemetry page. In that scenario , you will be able to adjust the slope-air speed by yourself and do not care about roll stability when you don't see the glider. 
Other solution is to use INAV on FC + gps + ASI , but in that case you will need a customize firmware for your FC , like you said to use elevation instead of throttle and keep slope-air speed relation in defined range. In other words , somebody has to change the code inside Inav.
Any way 5m glider is far out of my league , so good luck.

closed 2020-06-28 by stale[bot]