iNavFlight/inav #5580: Motor runaway - Full throttle when barely touching throttle2

BlueBird69, 2020-04-08:
Similar phenomena as reported by **tott** commented on 15 Apr 2019. 
I have folloved the steps diskussed in this thread #4613 without any luck. 

INAV 2.4.1. Multishot. 
FC Mateksys F772-SE with Matek GPS & COMPASS MODULE M8Q-5883. 
GEPRC 50A BLHELI_32 ESC. Calibrated. 
Configuration based on 7" Endurance with some small adjustments. 
Horus X10 with latest OpenTX (2.3.7) with TBS CRSF (with latest FW).

This drone did fly nicely with Betaflight.
I sure hope to get this sorted out and fly it with INAV and utilize return to home and hoover modes. 
I'll add Diff-All data for you for investigation. 

## Problem
Luckilly I pressed down the drone towards the ground at first test. Motors start softly at low throttle, but imeadetly when I add slightly increased thr., the motors accelerate and increases extreamly fast. Not flyable at all. It will rocket to the sky.  The RPM is increasing almost exponentially. Not 100% RPM imedeately, but increases very fast. 

## Steps to Reproduce
1. Turn on radio and drone
2. Wait for successfull boot-up and GPS lock. 
3. Disarm and gently add throttle. 
## Expected behavior
I expect that throttle gradually increase motor speed and keep relative constand rpm when throttle is steaddy. Just like all other drones fly :)

## Additional context
diff all
# version
# INAV/MATEKF722SE 2.4.0 Feb 11 2020 / 22:53:38 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]
# start the command batch
batch start
# reset configuration to default settings
defaults noreboot
# resources
# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000
# servo mix
# servo
# logic
# gf
# feature
feature -TX_PROF_SEL
feature -TELEMETRY
feature -AIRMODE
feature DYNAMIC_FILTERS
feature GPS
feature PWM_OUTPUT_ENABLE
# beeper
# map
# serial
serial 0 2 115200 115200 0 115200
serial 3 1024 115200 115200 0 115200
serial 5 4096 115200 115200 0 115200
# led
# color
# mode_color
# aux
aux 0 0 0 1300 2100
aux 1 1 1 900 1675
aux 2 9 1 1275 1675
aux 3 8 1 1700 2100
aux 4 11 2 1700 2100
aux 5 29 2 1700 2100
aux 6 30 6 1700 2100
aux 7 31 6 900 1300
aux 8 37 4 900 1300
aux 9 38 4 1300 1700
aux 10 32 5 900 1300
aux 11 33 5 1300 1700
aux 12 34 5 1700 2100
# adjrange
# rxrange
# temp_sensor
# wp
#wp 0 invalid
# osd_layout
osd_layout 1 0 24 1 V
osd_layout 1 1 1 12 V
osd_layout 1 2 0 0 V
osd_layout 1 3 8 6 V
osd_layout 1 4 8 6 V
osd_layout 1 7 13 12 V
osd_layout 1 10 12 10 V
osd_layout 1 11 2 11 V
osd_layout 1 12 1 10 V
osd_layout 1 13 3 8 V
osd_layout 1 14 23 2 V
osd_layout 1 15 22 8 V
osd_layout 1 20 7 14 V
osd_layout 1 21 18 14 V
osd_layout 1 22 15 4 V
osd_layout 1 23 13 3 V
osd_layout 1 24 12 2 V
osd_layout 1 25 26 5 V
osd_layout 1 34 10 1 V
osd_layout 1 37 1 14 V
osd_layout 1 38 1 13 V
osd_layout 1 86 1 1 V
osd_layout 1 97 10 13 V
osd_layout 1 105 15 9 V
osd_layout 1 106 2 2 V
osd_layout 2 0 24 1 V
osd_layout 2 1 1 12 V
osd_layout 2 2 0 0 V
osd_layout 2 3 8 6 V
osd_layout 2 4 8 6 V
osd_layout 2 13 3 7 V
osd_layout 2 14 23 2 V
osd_layout 2 15 22 7 V
osd_layout 2 20 7 14 V
osd_layout 2 21 17 14 V
osd_layout 2 22 15 4 V
osd_layout 2 24 12 2 V
osd_layout 2 34 10 1 V
osd_layout 2 38 1 13 V
# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 80
set gyro_stage2_lowpass_hz = 150
set acc_hardware = MPU6000
set acczero_x = 86
set acczero_y = -17
set acczero_z = -245
set accgain_x = 4074
set accgain_y = 4100
set accgain_z = 4040
set mag_hardware = QMC5883
set magzero_x = -992
set magzero_y = -528
set magzero_z = 81
set baro_hardware = BMP280
set pitot_hardware = NONE
set serialrx_provider = CRSF
set motor_pwm_rate = 2000
set motor_pwm_protocol = MULTISHOT
set throttle_idle =  5.000
set failsafe_procedure = RTH
set current_meter_scale = 100
set model_preview_type = 3
set applied_defaults = 2
set gps_sbas_mode = EGNOS
set nav_user_control_mode = CRUISE
set nav_mc_hover_thr = 1450
set osd_video_system = PAL
set osd_left_sidebar_scroll = ALTITUDE
set osd_right_sidebar_scroll = GROUND_SPEED
set osd_sidebar_scroll_arrows = ON
set name = CL7_iNAV
set vtx_band = 1
set vtx_channel = 7
set vtx_power = 1
set vtx_low_power_disarm = ON
set vtx_freq = 5745
# profile
profile 1
set mc_p_pitch = 45
set mc_i_pitch = 35
set mc_d_pitch = 24
set mc_i_roll = 35
set mc_d_roll = 24
set dterm_lpf_hz = 75
set use_dterm_fir_filter = OFF
set d_boost_factor =  1.500
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set rc_yaw_expo = 70
set roll_rate = 60
set pitch_rate = 60
set yaw_rate = 50
# profile
profile 2
set dterm_setpoint_weight =  0.500
set d_boost_factor =  1.100
set d_boost_max_at_acceleration =  5000.000
set d_boost_gyro_delta_lpf_hz = 55
set thr_mid = 0
set thr_expo = 73
# profile
profile 3
# battery_profile
battery_profile 1
set bat_cells = 4
set vbat_min_cell_voltage = 300
# battery_profile
battery_profile 2
# battery_profile
battery_profile 3
# restore original profile selection
profile 2
battery_profile 1
# save configuration
save
# 

I sure hope to get some thoughts and support on solving this issue. 
Kind regards,
Torgeir

digitalentity, 2020-04-08:
What flight  mode was active when you did the test?

BlueBird69, 2020-04-08:
> What flight mode was active when you did the test?

Hi Digitalentity! Angle mode all the time. 
Should I preferabley use anoghet mode=?
//Torgeir

digitalentity, 2020-04-08:
ANGLE should be fine, but I would do initial test in ACRO. Please check if motors behave correctly when controlled from the Output tab (take the propellers off first).

BlueBird69, 2020-04-09:
Hello Konstantin!
Thank you for your follow up questions.
I tested now in three modes:
1. Angle mode - Behaves as described earlier. With props on and carefully
pressing the drone down towards the table.  Just to get the props effekt in
the equation.
2. Horizon mode - Much like Angle Mode when only running motors on the
bench, with props on and holding the drone firm to the table.
3. Acro mode (or as described in the INAV WiKi - "Rate" Mode (or no mode
selected). Then the motors behave as expected and their RPM follows along
with the throttle position, and does not fluctuate or increase
"uncontrolled".

In INAV Configurater motor menu, the motors act like in "Acro" mode, with
RPM corrolateing with "throttle"-slider position.

Any suggestions?
I would like to use Angle mode with its benefits to stabelize the drone
while flyng, but as it acts now, I'm conserned it sky-rockets in no time.

Your thoughts and inputs to this is much apprechiated.
Best regards,
Torgeir

digitalentity, 2020-04-09:
Angle mode doesn't control throttle. The effect you are seeing is likely caused by the quad being not level (or FC alignment being wrong). FC is pushing hard to level itself and obviously fails since the quad doesn't move.

Can you do a blackbox recording of a test in ANGLE mode? This way we'll be able to look into  what's going on under the hood.

BlueBird69, 2020-04-09:
Hi,
The FC is mounted forrect direction and not flipped around. All axis
calibrated within INAV when the drone is horisongal on a very flat /level
surface.
Question: Might the barometer on the FC cause trouble when exposed to the
downwash from the props? I will ad some open-cell foam to cover it.

I have reduced Roll rate and Pitch rate to 500. Reduced Dterm setpoint
weight to 0.1%. TPA set to 0.1% (not sure if 0.0 is well accepted by INAV)
From this the Quad seem to behave well. When I hold it in the air and
increase the., it seem to correct angles and stabilize now. So next will be
to try it outside (when it stops tp rain/snow). Hopefully within a day or
two.

i'll keep you posted and will try to logg som data.
In the meantime your comments to the above mentioned question and
adjustments will be much of interest.

Br, Torgeir

digitalentity, 2020-04-09:
Very interested to see the results of your test.

ANGLE mode alone doesn't use anything but gyro & accelerometer sensors. Barometer and other sensors are used in more advanced modes (ALTHOLD etc).

Important thing to understand is that your perception of "level" and FC's perception of level might be different. For FC a 0.1deg error is "not level at all" and it will try to compensate. It should not be overreacting through, so I'm very curious about why your quad experiences this behavior in ANGLE.

BlueBird69, 2020-04-09:
Thanks for your reply! Very much appreciated!
Good to know, I'll need to look more thorough through the INAV
documentations.

According to what you describe regarding small deviation in level-position
and the FC detection/correction, this might describe the behaviour. I
just didn't expect motors to boost up so much, but at the same time I held
it firmly to the "flat" surface, so the quad/FC was kept more or less in
the same plane. Not letting the algorithm loop to achieve adjustments
according to it's corrections.

I'll insert a uSD cart into the FC and try to replicate the phenomenoa.
Then it might be logged as well. Then it might be documented. I'll see what
I can do before flyable weather turns up, as well (and try to keep all my
fingers unharmed, while testing :))

Another question - When configuring modes, will NAV PosHold or NAV RTH
override Angle mode or Horizon Modes? Or do I need to be careful and not
overlap these in the modes config tab (e.g combination of Angle Mode and
NAV RTH for the same switch position) ?

I'll see if I can get you a blackbox logg soon.
Thanks!
Torgeir

BlueBird69, 2020-04-10:
Case closed!

closed 2020-04-10 by BlueBird69