iNavFlight/inav #5582: Inav launch mode won’t activate using mode other than manual

lightning10811, 2020-04-10:
I have matek 405wing, inav 2.3.  
Problem: auto launch will only activate and work when manual mode selected.  I select launch, then mode other than manual (RTH for instance) then arm, then set throttle.  The board will not proceed to auto launch.  But if I leave it in manual mode, set launch, arm board, set throttle then it works.  I have tried setting RTH (or horizon mode), then go through the above process and this doesn’t work either.  Seems as if auto launch only works when manual is selected.

Any thoughts?

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

digitalentity, 2020-04-10:
Please define "doesn't work". Board doesn't arm? Also provide config dump.

lightning10811, 2020-04-11:
Thanks for the follow up!  Cli dump is below.  By “doesn’t work” I mean that I throw the plane and the motor doesn’t spin and the plane hits the ground.  To better test this, I turned off the command “don’t spin motor when armed” so I could know if it was armed.  Then, in manual mode (channel 6) I select launch (channel 5),  arm  (channel 7) the motor then spins at idle speed even if I move throttle up to max which means auto launch is ready.  But, If I start this sequence in any other mode, the motor doesn’t spin at idle speed showing it’s not armed and launch not ready.  And, if I sling the plane to see if it activates the motor (to test if it’s  in  Launch mode) the motor doesn’t spin.

Please help.  Thanks again.

Here’s the cli dump:

 dump

# version
# INAV/MATEKF405SE 2.3.0 Nov 24 2019 / 11:01:22 (d0bdd38c0)
# GCC-8.2.1 20181213 (release) [gcc-8-00 100
servo 1100
servo 15 1000 2logic 3 0 0 0 0 0 00
logic 8 0 0 0 0 00
logic 13 0 0 0 0  0 0
gf 2 0 0 0 0 0
# feature
featureOR_STOP
feature -DYe -3D
feature -RSSIe -AIRMODE
feature 
feature -FW_LAUNCHfeature PWM_OUTPUT_ET
beeper RX_LOST_LAOW
beeper GPS_STATUS
beeper DISARM_REPPEN
beeper CAM_CONN00 115200 0 115200
 115200
serial 4 1 # led
led 0 0,0::C::C:0
led 7 0,0::C::C:0
led 14 0,0::C:ed 20 0,0::C:0
led led 27 0,0::C:0
ledor 2 0,0,255
color 180,0,255
color 9 2 14 0,0,0
color 15  13
mode_color 0 4 ode_color 1 4 10
mocolor 2 4 10
mode_cor 3 4 10
mode_colo 4 4 10
mode_color 0 1
mode_color 0 1 de_color 6 5 2
mode 6 10 0

# aux
auaux 5 0 0 900 900
a 0 900 900
aux 12 0aux 17 0 0 900 900
0 0 900 900 0 0
adj
adjrange 6 0 0 9000
adjrange 10 0 0 9ange 2 1000 2000
rx 0 0 0 0 0 
temp_setemp_sensor 7 0 0 0 ge 5 0 0 900 900 0 0 0
wp 10 0 0 0 0 0  0 0 0 0 0 0
wp 16 1 0 0 0 0 0 0
wp 220 0
wp 28 0 0 0 0 00 0 0 0
wp 34 0 0 0 0
wp 40 0 0 0 0 0 0 0 0 0
wp 47 0 0 0 0
wp 53 0 0 0 0 0  0 0
wp 27 0 0 0 0 _layout 0 3 8 6 V
ot 0 7 13 10 V
osd_l3 6 V
osd_layout 0 1 H
osd_layout 0 18t 0 22 13 11 V
osd_d_layout 0 27 3 5 H
ut 0 32 12 1 H
osd_yout 0 37 1 6 H
[... 138 lines cut ...]
reakpoint = 1500
set fw_tpa_time_constant = 0
set rc_expo = 60
set rc_yaw_expo = 60
set roll_rate = 30
set pitch_rate = 20
set yaw_rate = 10
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 5000
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

# end the command batch
batch end

digitalentity, 2020-04-11:
Dump  is garbled unfortunately.  Might indicate hardware or driver problem.
Please check via OSD that the plane arms. Looks like you simply don't have a GPS fix and INAV doesn't allow you to arm. OSD will also indicate the reason for not arming.

lightning10811, 2020-04-11:
I used the OSD as you said which gives message of "armed" and "autolaunch" when in manual mode. When in any other mode, it just doesn't arm and no messages or error message is given.   

Should I update to iNav 2.4?

Here's the dump again listed below.  The first one was from speedy bee.  This one is from my computer and looks much better.  

# dump

# version
# INAV/MATEKF405SE 2.3.0 Nov 24 2019 / 11:01:22 (d0bdd38c0)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0 1.000 0.000 0.000 0.000
mmix 1 1.000 0.000 0.000 0.000

# servo mix
smix reset

smix 0 3 0 -100 0 -1
smix 1 4 0 -100 0 -1
smix 2 2 1 -100 0 -1

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1497 -100
servo 3 1000 2000 1458 100
servo 4 1000 2000 1448 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# logic
logic 0 0 0 0 0 0 0 0
logic 1 0 0 0 0 0 0 0
logic 2 0 0 0 0 0 0 0
logic 3 0 0 0 0 0 0 0
logic 4 0 0 0 0 0 0 0
logic 5 0 0 0 0 0 0 0
logic 6 0 0 0 0 0 0 0
logic 7 0 0 0 0 0 0 0
logic 8 0 0 0 0 0 0 0
logic 9 0 0 0 0 0 0 0
logic 10 0 0 0 0 0 0 0
logic 11 0 0 0 0 0 0 0
logic 12 0 0 0 0 0 0 0
logic 13 0 0 0 0 0 0 0
logic 14 0 0 0 0 0 0 0
logic 15 0 0 0 0 0 0 0

# gf
gf 0 0 0 0 0 0 0
gf 1 0 0 0 0 0 0
[... 1133 lines cut ...]
set roll_rate = 30
set pitch_rate = 20
set yaw_rate = 10
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 400
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 5000
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

# end the command batch
batch end

closed 2020-06-26 by stale[bot]