iNavFlight/inav #5583: configurator and osd select the wrong frequents for vtx (looklike its swapped)

modwadd, 2020-04-10:
## Current Behavior
FC: MatekF765
FTX: echine tx805 v2.0
inav: 2.4.0
If I chous Band A channel **1** the transmitter switxch tol Band A channel **8**!
If I chouse Band A channel **8** the transmitter switxch tol Band A channel **1**!
## Steps to Reproduce

1. Setup FC an xtx
2.Powert up and tune vrx to get picture (including chosen channel displayed in osd))
3.Set channel via configurator or OSD, channel is switche but on wron channel.
4.

## Expected behavior

Channel which ist choosen in OSD or configurator is choosen (send to ) vtx to change to
## Suggested solution(s)

look like ther is an index issue on the channel table.
the table in vtx_string.c is correct.
## Additional context

---

will follow
- FC Board name and vendor:
- INAV version string:

modwadd, 2020-04-10:
[INAV_cli_20200408_212419_f765_v2ok.txt](https://github.com/iNavFlight/inav/files/4462847/INAV_cli_20200408_212419_f765_v2ok.txt)

DzikuVx, 2020-04-14:
Are you sure the problem is not on the VTX side?

modwadd, 2020-04-15:
Yes with the MATEKF765 the same vtx is working (neerly) corret, at least it is switching channles.
From my poinit of view it is an port setting (push pull, or pull down or ..) or hardware issue on the Wing FC.
BR
Kai

modwadd, 2020-04-26:
Hello,
I figured out the issue now.
With a secont SA VTX the issue is the same.
But its not a bug in the iNav its in my Quantum 5.8 Diversety rceiver, it displais for band A the channels wrong. (channel 8 to 1 not 1 to 8)

BR

closed 2020-04-26 by modwadd