iNavFlight/inav #5584: Cannot set 10hz update mode

wx4cb, 2020-04-10:
HW: latest release version running on a new matek F722, no soft serial, and gps on UART3 (115k2 bps)

Problemm: even if setting update rate in ucenter to 10hz, my M8N module will not go to 10hz in inav. the FC always resets it back to 5hz.

if i load it and save it in ucenter disconnect/close ucenter, and reconnect via gpspassthrough after powering down just the gps ucenter sees it as being at 10hz.

any ideas? I know there used to be an issue, but i thought we were doing version checking now.

What can I check in inav cli to see what's going on if possible?

```
# status
System Uptime: 236 seconds
Current Time: 2020-04-10T14:34:33.487-04:00
Voltage: 0.20V (1S battery - NOT PRESENT)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: Startup failed
Stack size: 6144, Stack address: 0x20010000, Heap available: 1716
I2C Errors: 45, config size: 5811, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 4, used = none
System load: 13, cycle time: 258, PID rate: 3875, RX rate: 49, System rate: 9
Arming disabled flags: RX CLI AUTOTRIM

# diff

# version
# INAV/MATEKF722SE 2.4.0 Feb 11 2020 / 22:53:38 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# start the command batch
batch start

# resources

# mixer
mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix 0 3 0 -50 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 50 0 -1
smix 3 4 1 50 0 -1

# servo
servo 3 1000 2000 1551 100
servo 4 1000 2000 1546 -100

# logic

# gf

# feature
feature -TX_PROF_SEL
feature MOTOR_STOP
feature DYNAMIC_FILTERS
feature GPS
feature LED_STRIP
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial
serial 0 0 115200 115200 115200 115200
serial 1 0 115200 57600 0 115200
serial 2 2 115200 115200 0 115200
serial 3 256 57600 115200 57600 115200
serial 5 64 115200 115200 0 115200

# led
led 0 0,0:WU:CTB:1
led 1 0,1:WU:CTB:1
led 2 0,2:WU:CTB:1
led 3 0,3:WU:CTO:5
led 4 0,4:WU:CTO:5
led 5 0,5:WU:CTO:5
led 6 0,6:WU:CTO:5
led 7 0,7:WU:CTO:5
led 8 0,8:WU:CTO:5
led 9 0,9:WU:CTO:5
led 10 0,10:WU:CTO:5
led 11 0,11:WU:CTO:5
led 12 0,12:WU:CTB:1
led 13 0,13:WU:CTB:1
led 14 0,14:WU:CTB:1
led 15 15,0:EU:CTB:1
led 16 15,1:EU:CTB:1
led 17 15,2:EU:CTB:1
led 18 15,3:EU:CTO:2
led 19 15,4:EU:CTO:2
led 20 15,5:EU:CTO:2
led 21 15,6:EU:CTO:2
led 22 15,7:EU:CTO:2
led 23 15,8:EU:CTO:2
led 24 15,9:EU:CTO:2
led 25 15,10:EU:CTO:2
led 26 15,11:EU:CTB:1
led 27 15,12:EU:CTB:1
led 28 15,13:EU:CTB:1

# color

# mode_color

# aux
aux 0 0 0 1925 2100
aux 1 1 1 1125 1275
aux 2 3 1 1350 1450
aux 3 3 1 1750 1850
aux 4 9 1 1750 1850
aux 5 8 1 1925 2075
aux 6 35 1 1350 1450
aux 7 10 1 900 1075
aux 8 27 3 1450 1550
aux 9 28 3 1925 2100
aux 10 32 2 1450 1550
aux 11 33 2 1925 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 26 0 V
osd_layout 0 1 1 0 V
osd_layout 0 3 8 6 H
osd_layout 0 4 8 6 H
osd_layout 0 7 17 14 V
osd_layout 0 9 23 1 H
osd_layout 0 11 7 0 V
osd_layout 0 12 1 1 V
osd_layout 0 13 19 1 V
osd_layout 0 14 23 13 V
osd_layout 0 15 7 1 V
osd_layout 0 20 1 14 H
osd_layout 0 21 1 13 H
osd_layout 0 22 15 14 V
osd_layout 0 23 19 0 V
osd_layout 0 24 13 1 V
osd_layout 0 26 12 11 H
osd_layout 0 28 23 14 V
osd_layout 0 29 23 10 H
osd_layout 0 30 1 4 V
osd_layout 0 31 16 13 V
osd_layout 0 33 24 1 V
osd_layout 0 34 11 1 H
osd_layout 0 40 12 0 V
osd_layout 0 41 25 12 H
osd_layout 0 42 21 12 H
osd_layout 0 45 0 0 V
osd_layout 0 46 23 14 H
osd_layout 0 47 16 14 H
osd_layout 0 50 7 14 V
osd_layout 0 85 8 14 H
osd_layout 0 86 23 13 H
osd_layout 0 87 1 13 H
osd_layout 0 88 2 7 H
osd_layout 0 89 14 1 H
osd_layout 0 97 1 13 V

# master
set looptime = 250
set gyro_hardware_lpf = 256HZ
set acc_hardware = MPU6000
set acczero_x = 61
set acczero_y = -33
set acczero_z = -387
set accgain_x = 4098
set accgain_y = 4082
set accgain_z = 4021
set mag_hardware = NONE
set mag_declination = -600
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 12
set serialrx_provider = CRSF
set max_throttle = 2000
set motor_pwm_rate = 2000
set motor_pwm_protocol = MULTISHOT
set motor_poles = 10
set failsafe_procedure = RTH
set align_board_pitch = 36
set current_meter_scale = 250
set platform_type = AIRPLANE
set model_preview_type = 8
set servo_pwm_rate = 60
set small_angle = 180
set applied_defaults = 1
set nav_rth_climb_first = OFF
set nav_rth_allow_landing = FS_ONLY
set nav_rth_altitude = 6096
set nav_fw_cruise_thr = 1450
set nav_fw_max_thr = 2000
set nav_fw_climb_angle = 25
set nav_fw_loiter_radius = 3000
set osd_video_system = PAL
set osd_units = IMPERIAL
set osd_rssi_alarm = 0
set osd_time_alarm = 0
set osd_alt_alarm = 0
set osd_dist_alarm = 0
set osd_neg_alt_alarm = 0
set osd_crosshairs_style = AIRCRAFT
set osd_coordinate_digits = 8
set osd_plus_code_digits = 10
set name = WX4CB-Harrier
set tz_offset = -240

# profile
profile 1

set fw_ff_pitch = 65
set fw_ff_roll = 65
set fw_ff_yaw = 65
set fw_p_level = 40
set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 600
set rc_expo = 0
set rc_yaw_expo = 0
set roll_rate = 60
set pitch_rate = 60
set yaw_rate = 60
set manual_rc_expo = 0
set manual_rc_yaw_expo = 0

# battery_profile
battery_profile 1

# end the command batch
batch end

# 
```

wx4cb, 2020-04-10:
[image]

stronnag, 2020-04-10:
` set gps_provider = UBLOX7`

stronnag, 2020-04-10:
for posterity #5585

wx4cb, 2020-04-10:
> `set gps_provider = UBLOX7`

i knew it had to be something stupid. as I knew this was already capable of 10hz in the previous incarnation.

closed 2020-04-10 by wx4cb