iNavFlight/inav #5586: Acro Trainer

TIMO88D, 2020-04-11:
## Current Behavior

Feature not present 
## Desired Behavior

When in Acro Trainer mode the craft will behave as it would in Acro Mode but will prevent itself going beyond a set angle or will return itself to that angle if forced beyond it (by a gust). This is useful when learning to fly Acro or when flying in gusty conditions. This is different to angle mode as within the angle limit it will not self stabilise.
## Suggested Solution

Take the feature already present in betaflight and implement in INAV.  
## Who does this impact? Who is this for?

Useful for beginners, people flying in gusty conditions or crosswinds, or planes that have wing over issues. 
## Additional context

Feature presesent in betaflight since 3.4. Not sure if you can just copy the code over.

TIMO88D, 2020-06-10:
Would still love this feature.

elielmarcos, 2021-05-11:
I would also love to have the Acro Trainer mode added to iNav.

BetaFlight already has this in many past versions, but I prefer to use the features of iNav!

I am a beginner on the Acro flight, I have always piloted in Angle mode and would not like to destroy my quad on the first attempt;)

RecifeStargazer, 2021-11-10:
it would be nice to have It.

psfales, 2024-12-15:
I would also like to see this.   I wondered if Inav had this capability, and the search led me here.

I do flight instruction, and I think it would be great if the students could learn in acro mode (so that they don't learn bad habits from angle mode) but it would still prevent them from flipping plane or otherwise going too crazy.

sensei-hacker, 2024-12-16:
> I would also like to see this. I wondered if Inav had this capability, and the search led me here.
>
> I do flight instruction, and I think it would be great if the students could learn in acro mode [...]

It would be fairly straightforward to set this up in the Programming tab and try it out, see how it actually works out in practice.

Greater than, 40, flight - Pitch
...

psfales, 2024-12-16:
The programming idea is intriguing, but being fairly new to inav, I'm not sure what action should be taken when the condition is triggered.   (I'm probably more interested in limiting roll, but pitch would be good too)

sensei-hacker, 2024-12-16:
"I'm not sure what action should be taken when the condition is triggered"

Neither am I. The concept is intriguing; the details would need some thought. Engaging Loiter mode for the next 10 seconds might get them out of trouble. Or might get them into trouble.

psfales, 2024-12-16:
Also  wondering about engaging angle mode for a few seconds.   (I had thought of angle mode before, but wasn't sure what condition could be used to turn it off.  I didn't know that a time delay was possible)

I'm very new to inav programming.  Can you suggest how I would implement the action of entering angle (or loiter) for a few seconds and the returning to acro mode?

psfales, 2024-12-16:
I was watching a video today and saw a reference to "Self Leveling Mode. "  That might be another option

sensei-hacker, 2024-12-17:
> I was watching a video today and saw a reference to "Self Leveling Mode. " That might be another option

There are several modes, including Angle and horizon which are "self-leveling".

sensei-hacker, 2024-12-17:
> I'm very new to inav programming. Can you suggest how I would implement the action of entering angle (or loiter) for a few seconds and the returning to acro mode?

In the Programming tab, there isn't a direct control of flight mode, so what you would do is override the RC channel that controls mode. This could even be a channel that isn't set up on the radio, of you wanted. So something like this, of mode is on channel 8:

Override RC channel, value, 8, 1800

That would cause it channel 8 to be set to 1800 for as long as the bank angle is too high. Then set it to do Angle mode or loiter or whatever when channel 8 is 1800.

To l use a timed, do you want to the timer start when the bank angle is first exceeded, or should the timer begin when the angle is restored, going to Angle mode and staying there for a couple seconds AFTER the condition is rectified?

psfales, 2024-12-17:
I'm not sure when I'll be able to try this, as between the holiday season and the weather the opportunities for outdoor flying are limited, but I will experiment with this.    I'm thinking that as soon as the bank angle is exceeded it would pop into angle mode and would only need to stay there for a couple of seconds.     Probably either the start of angle mode, or rectifying the angle would work (as I would expect those to be almost simultaneous), but I think would wait for the condition to be rectified just in case the student was going crazy on the sticks!

Thanks for all the suggestions!

bd4, 2025-03-23:
Re programming this mode: How about when angle limit is exceeded, and when rc channel is < center in the same direction, it overrides channel to 0? This would be tedious and require a bunch of rules for each axis and each direction, but maybe it would work. In any case I would love to have a proper acro trainer mode, for the purpose of training beginners.