iNavFlight/inav #5594: Omnibus F4V3 Broken support for PPM receiver when softserial target selected

robertha64, 2020-04-14:
## Current Behavior
Case 1: When flashed OmnibusF4V3 target to F4V3 board, PPM is working on SBUS/PPM pin.
Case2: When flashed the same board with Omnibus F4V3 S5S6_SS, Omnibus F4V3 S5_S6_2SS or Omnibus F4V3 S6_SS the PPM input is not working.

resource output case1:
 resource
IO:
----------------------
A00: FREE 
A01: FREE 
A02: MOTOR4 OUT
A03: MOTOR3 OUT
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: FREE 
A09: SERIAL1 UART TX
A10: SERIAL1 UART RX
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: OSD CS
B00: MOTOR1 OUT
B01: MOTOR2 OUT
B02: FREE 
B03: BARO CS
B04: BEEPER OUT
B05: LED1 OUT
B06: FREE 
B07: SDCARD IN
B08: PPM IN
B09: FREE 
B10: I2C2 SCL
B11: I2C2 SDA
B12: SDCARD CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: FREE 
C01: FREE 
C02: ADC CH2
C03: FREE 
C04: MPU EXTI
C05: FREE 
C06: SOFTSERIAL1 UART TX
C07: FREE 
C08: INVERTER OUT
C09: INVERTER OUT
C10: SPI3 SCK
C11: SPI3 MISO
C12: SPI3 MOSI
C13: FREE 
C14: FREE 
C15: FREE 
D00: FREE 
D01: FREE 
D02: FREE 
D03: FREE 
D04: FREE 
D05: FREE 
D06: FREE 
D07: FREE 
D08: FREE 
D09: FREE 
D10: FREE 
D11: FREE 
D12: FREE 
D13: FREE 
D14: FREE 
D15: FREE 

# 

resource output case2:
resource
IO:
----------------------
A00: FREE 
A01: SOFTSERIAL1 UART TX/RX
A02: MOTOR4 OUT
A03: MOTOR3 OUT
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: SOFTSERIAL2 UART TX/RX
A09: SERIAL1 UART TX
A10: SERIAL1 UART RX
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: OSD CS
B00: MOTOR1 OUT
B01: MOTOR2 OUT
B02: FREE 
B03: BARO CS
B04: BEEPER OUT
B05: LED1 OUT
B06: FREE 
B07: SDCARD IN
B08: FREE 
B09: FREE 
B10: I2C2 SCL
B11: I2C2 SDA
B12: SDCARD CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: FREE 
C01: FREE 
C02: ADC CH2
C03: FREE 
C04: MPU EXTI
C05: FREE 
C06: FREE 
C07: FREE 
C08: INVERTER OUT
C09: INVERTER OUT
C10: SPI3 SCK
C11: SPI3 MISO
C12: SPI3 MOSI
C13: FREE 
C14: FREE 
C15: FREE 
D00: FREE 
D01: FREE 
D02: FREE 
D03: FREE 
D04: FREE 
D05: FREE 
D06: FREE 
D07: FREE 
D08: FREE 
D09: FREE 
D10: FREE 
D11: FREE 
D12: FREE 
D13: FREE 
D14: FREE 
D15: FREE 

# 

## Steps to Reproduce

1. Program Omnibus F4V3 board (will work the same way also on Omnibus F4Pro board) with Omnibus F4V3 FW - check PPM - should work 
2. Program Omnibus F4V3 or F4PRO board with F4V3 S5S6_SS or F4V3 S5_S6_2SS FW - ckeck PPM - will not work
3.
4.

## Expected behavior
There is no reason why softserial on A1(M5) and A8(M6) will disable PPM on B08
## Suggested solution(s)
Revert latest changes to target.c for Omnibus F4, as I believe it did work correctly some time ago 

## Additional context

---

- FC Board name and vendor:
Omnibus F4V3 and Omnibus F4PRO both from Banggood
- INAV version string:
 
# INAV/OMNIBUSF4V3_S6_SS 2.2.1 Jul  3 2019 / 22:15:59 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

robertha64, 2020-04-18:
I think is broken because of this change, because when I build user fw with change reverted, PPM is working.
[image]

tiriad, 2020-04-19:
> I think is broken because of this change, because when I build user fw with change reverted, PPM is working.
> [image]

I tried the variant Omnibus F4v3 s5 s6  2ss and detects ppm in pb8, that change you indicate was made so that all variants Omnibus f4v3 were compatible with ppm and not make the code redundant, try to test if there is any configuration that you have not taken into account
target.h
[image]
target.c
[image]
[image]

robertha64, 2020-04-23:
I did try 2.4 now (S4S6_SS), PPM is there but no softserial on A01 and/or A08.

Whats wrong ? 

[image]

[image]

digitalentity, 2020-04-23:
Too many different OMNIBUS F4 clones I guess...

tiriad, 2020-04-23:
> I did try 2.4 now (S4S6_SS), PPM is there but no softserial on A01 and/or A08.
> 
> Whats wrong ?
> 
> [image]
> 
> [image]

@robertha64 I tried again, the softserial will appear when you activate a peripheral
[image]

> Too many different OMNIBUS F4 clones I guess... I'll have a look

@digitalentity I think it's not a code problem, it's a firmware configuration problem

robertha64, 2020-04-24:
I believe I did activate SmartPort for Softserial1 but no sign of softserial in resources, I will try again later.

tiriad, 2020-04-24:
> I believe I did activate SmartPort for Softserial1 but no sign of softserial in resources, I will try again later.

For set SmartPort also remember to activate first enable Telemetry (feature telemetry)
https://github.com/iNavFlight/inav/blob/master/docs/Telemetry.md

robertha64, 2020-04-24:
Good catch, SmartPort was correctly setup on softserial, but telemetry not activated, now is fine. Closing.
[image]

closed 2020-04-24 by robertha64