iNavFlight/inav #5595: Add Prearm mode

Thunfisch2, 2020-04-14:
## Current Behavior
The vehicle will arm after entering the arm mode.

## Desired Behavior
The arming process should be staged. The arm switch will not work before a pre-arm switch has been activated. This will help to prevent accidentally arming the vehicle.    

## Suggested Solution
Adding a new pre-arm mode to the software. The arm mode must have a dependence on the pre-arm mode. After arming, the vehicle should stay armed after deactivating the pre-arm mode. This is helpful to get the possibility to put the pre-arm mode to a momentary switch. 

## Who does this impact? Who is this for?
All users who want to add a piece of security to the hobby. Rotating propellers can be very dangerous.

digitalentity, 2020-04-15:
This can be implemented on your radio TX if this is really needed. I don't think this adds more security, but we may consider this if the demand is significant.

MrD-RC, 2020-04-20:
Personally, I don’t think it’s needed. This can be set up in the Tx if your really want it. INAV also has safety conditions that must be met before you can arm.

CipherSoftRepo, 2020-07-08:
Please add it. A lot of people are use to it. I just installed inav now, and it's really only missing that :)

jerryt1, 2020-09-29:
I almost cut myself (or worse) today as I was attempting to disconnect the battery from my 7 inch drone after a flight.  While having difficulty removing the battery plug, I accidentally hit the Arm button on my controller, which I was holding because I didn't want to place it into the mud beneath me.  The props started up and I just dropped the drone into the mud rather than chance losing a digit.  My BetaFlight drones all use the push button on my controller for PreArm and I depend on it, but INAV doesn't yet support that.  Please, Please PLEASE add this feature!  It is likely an easy coding to implement and greatly improves safety.  Thanks much!

giacomo892, 2020-09-30:
I always rise the throttle stick before getting close to my quads when I have the radio with me. Easy and effective fix preventing arming

jerryt1, 2020-09-30:
That's an excellent suggestion!  I don't feel it replaces the much greater safety provided by adding the PreArm Mode to INAV, but it is another good method to enhance our safety. Having PreArm is an even more when its my 10 inch quad!  I carry my controller when going after a drone that has come down unexpectedly, such as because I caught a small branch I didn't see until the last moment or a failure of a part, such as a motor.  Once a quad is lost, we're often in for a search.  With my controller, I can turn the beeper on every time I need a new "fix" and display the latitude and longitude on the screen to find the search area, etc.

Another suggestion made was to implement this function in the controller.  I'm and engineer and code developer, I can do that.  But it's certainly not going to be as usable to newer or non-technical pilots.  Conversely, everyone building/configuring quads must learn how to add and modify Modes, so adding PreArm makes the safety feature easily available to all pilots.

I'm going to try to remember your great suggestion - thanks!

Redeque, 2020-10-16:
Please implement this. With software pre-arm you can use a button instead of a switch. Very handy with like the TBS Tango 2 that doesnt have a lot of switches.

Thunfisch2, 2020-10-26:
I am using a DJI remote and I have no possibility to set a pre arm on the remote and I have to rely on a software solution.

chr0mcom, 2020-11-22:
really need this! plz implement it

harry1453, 2021-02-03:
Just opened a PR for this. I think it's a really important safety feature. Please could people who want this feature test out my branch?

chr0mcom, 2021-02-03:
do have a hex file for testing?

avsaase, 2021-02-03:
@harry1453 Can you explain how this feature works? I'm interested to test it but I'm not familiar with the concept of pre-arm.

harry1453, 2021-02-03:
@avsaase Basically, if you have a prearm mode setup, you need to arm in two stages:

1. Activate prearm (normally a button or momentary switch)
2. Activate arm

If you don't arm in that order (eg. you arm then press prearm, or arm without prearm) the aircraft won't arm. It's a safety feature to prevent accidentally bumping the arm switch and chopping your fingers off.

harry1453, 2021-02-03:
@chr0mcom hold on, I'll prepare them now that I've made some changes to the PR

harry1453, 2021-02-04:
@chr0mcom Actually, you can get them from the CI artifacts:

https://github.com/iNavFlight/inav/suites/1962527221/artifacts/38930241

chr0mcom, 2021-02-08:
im looking for a inav configurator which is working with the 2.7 firmware... :/

stronnag, 2021-02-09:
> im looking for a inav configurator which is working with the 2.7 firmware... :/

http://seyrsnys.myzen.co.uk/inav-configurator-next/

harry1453, 2021-02-28:
#6562 Has been merged, I think this can be closed.

closed 2021-03-01 by digitalentity