iNavFlight/inav #5600: Fixed Wing: Disable arming safety while plane is flying

PaulWieland, 2020-04-19:
## Current Behavior
If you disarm a plane mid flight, iNav will not allow you easily rearm. 

There is an option to prevent disarm unless throttle is zero (`disarm_kill_switch`), however - I just spent 2 hours searching for a friends plane in the woods because he forgot to set `no pulses` as his failsafe mode on his receiver.

The failsafe briefly caused his throttle to go to zero and for the switch to disarm. He didn't even realize what had happened until it was too late.

## Desired Behavior
Allow immediate rearming while flying.

As far as I know, betaflight & KISS have no arming safety, so this feature is quite unexpected for those of us coming from that world.

## Suggested Solution

1. Determine that the plane has already been armed and launched.
1. Check the sensors to determine if the plane is in flight.
1. * Check speed first (if equipped with GPS or Pitot). If moving forward, allow instant rearming.
    * If no GPS or Pitot, perhaps flight can be detected by the feedback the Gyro & Accelerometer receives. This must be different than if someone is holding the plane or if it is resting on the ground. 

## Who does this impact? Who is this for?
I believe this would benefit all fixed wing pilots.
## Additional context
I crashed 3 times personally from accidentally disarming just after take off until I discovered the `disarm_kill_switch` setting.

My friend's accident occurred because his failsafe setting on the receiver was setup incorrectly - however - without iNav's arming safety the situation would have been recoverable as the signal loss was temporary.

DzikuVx, 2020-04-24:
With INAV 2.4 it will be possible to override extra arming safety with logic conditions and global functions. Painless360 made a video about it https://www.youtube.com/watch?v=bWktM8WBV6k

INAV 2.5 will bring a user inteface to the functionality

closed 2020-04-24 by DzikuVx

PaulWieland, 2020-04-24:
@DzikuVx Thanks for the link!

Please consider that only people who have crashed after accidentally disarming are going to find this work around. It would be best if it could somehow be a default behavior to prevent crashed/lost models.

Keep up the good work and thanks again!

caiuspetronius, 2021-02-15:
With iNAV 2.6 there is not global functions. Instead, you go to the Programming tab, set your logic 0 switch to Greater Than - Flight - 10 s - Always, then in the next logic switch choose Override Safety Checks Arming - Logic 0

trailx, 2023-07-19:
So I just had this happen to me, and I went looking for why this happened.  If you're curious:
https://youtu.be/nzrGLPsU8Ls
There is no reason to allow a fixed wing to disarm on a failsafe, and then not allow it to rearm.  I'm not sure if this should be a new feature request, but I feel like this needs to be discussed.

Ultimately this is on me, as I didn't set up my failsafe correctly on my rx, but... this is an easily overlooked item that is very difficult to test.

Would anyone actually want their plane to disarm midflight, or disarm immediately after a failsafe?

MrD-RC, 2023-07-19:
Sorry, but failsafe is very easy to test. You turn off your radio and see if the parachute in configurator turns red. Then you turn on your radio and see if the parachute goes back to grey. If it does those 2 things. Your receiver is sending the correct failsafe message to the flight controller. Then you just need to setup what happens when you failsafe. 

Ultimately, this is something that should be in your pre-maiden checklist. INAV can’t be responsible for bad setups.

trailx, 2023-07-19:
I'm not saying that INAV is responsible here.  But it seems like a good idea for a future change.

The test you mentioned sounds super simple but I'll be honest, I dont think that's been on my radar to do.  Seems obvious now but I obviously thought I had it all set correctly.  How would that test have illuminated the incorrectly set disarm switch?  Would the parachute illuminate then just extinguish itself?

Is there a premaiden checklist written out that includes this?  I really thought I checked everything, but missed this simple test.

PaulWieland, 2023-07-19:
Bottom line is that inav's default config is such that a plane can be lost because inav wont allow the pilot to rearm in certain situations. Anyone coming from many other flight control firmwares or free control will be burned by this.

MrD-RC, 2023-07-19:
The disarm switch has nothing to do with failsafe. When the failsafe occurs, nothing is received from the transmitter. When set correctly, failsafe should not put switches in to set positions. It should just tell the FC that it is no longer receiving a signal. When the link is reestablished, the switches will be in the positions they were left in. A correctly setup failsafe will not cause a disarm. 

Which RC link are you using?

[This](https://www.mrd-rc.com/tutorials-tools-and-testing/flight-controller-therapy/inav-pre-maiden-checklist-a-helpful-reminder-and-saver-of-foam/)  is a fixed wing checklist. It still seems fine to me on 6.1.

MrD-RC, 2023-07-19:
> Bottom line is that inav's default config is such that a plane can be lost because inav wont allow the pilot to rearm in certain situations. Anyone coming from many other flight control firmwares or free control will be burned by this.

Not true. You can always rearm, especially with a fixed wing. Fixed wings can glide, and you still have control over the control surfaces. I have plenty of time. Just lower the throttle, switch out of a nav mode, rearm, raise the throttle, fly. If you have launch mode permanently enabled, you may need to wiggle the sticks. If you have autotrim or autotune enabled, you’ll have to disable it. Any blockers will show up in the system messages. Or, if that’s too difficult, there’s this https://youtu.be/ndxtVzC7LHA

trailx, 2023-07-20:
I don't know if you saw my video but when you're first confronted with a rearm issue you don't have much time to decide what nav mode you're in.  I was in Cruise mode when my plane failsafed, and I wasn't until after I crashed that I realized INAV considers that a navigation mode.  Getting back to the first point of this thread maybe - that would have also prevented my crash, if I simply could have rearmed.

Based on your response it sounds like there are a bunch of other restrictions that may not be met to rearm while flying.  None of them seem intuitive, and I literally had 9 seconds between realizing I was going down to crashing.  Hardly enough time to decide what was preventing me from rearming while my brain is panicking.

MrD-RC, 2023-07-20:
@trailx that didn’t look like a failsafe. You would have seen a system message saying RX LOSS and the mode would switch to !FS! if it were a failsafe. I guess this is what you were meaning by failsafe not being set up correctly. I can only guess than instead of sending *no pulses* or *hold*, you are sending switch positions when a failsafe occurs. Which should only ever be used as a last resort. TBH, I’m surprised DragonLink had any issues so close. 

With regards to re-arming. I know it’s easier said than done. But the key is not to panic. On the OSD it said **DISABLE NAVIGATION TO ARM**. If you had switched to manual, acro, angle, or horizon. It would have re-armed. So you didn’t need to decide what needs changing. INAV reminds you. Just check the system messages on screen. 

I think it would be nice if there was a short window in which you could rearm; if the model still thinks it’s flying. There were some extra checks added in 6.0 that can make this more likely. So maybe it will exist in future versions.

But ultimately, I think you really need to get that failsafe sorted out. You shouldn’t be using set switch positions unless you absolutely have to. If DragonLink is using SBUS. It should be able to set a failsafe bit to alert the flight controller. I’m not saying that as an excuse. But more so that you have a better setup going forward. I’d also maybe check your antennas or outputs power. DragonLink should have found that distance as easy as if it were next to you.

trailx, 2023-07-20:
Thanks for the advice.  I will look into that in more detail going forward.  I have never had a failsafe before this flight, and there may be a couple of contributing factors to it's occurance.  I may have been using a bad antenna, I'll need to test that more.  Also right after the crash, an E-2 flew over the river at 500 ft, I suspect their radar may have periodically drowned me out.  You'll note that earlier in the flight the RSSI also dropped out.  In retrospect, I should have returned home at that point instead of pressing on.

Certainly a learning experience.  I like the idea of the short window to allow rearming when there's an accidental disarm while flying.