iNavFlight/inav #5604: Configurator shows compass calibrated and navigation safe when GPS and Magnetometer are not connected

kelleysislander, 2020-04-19:
## Current Behavior

With the GPS and magnetometer disconnected from the board, the Setup page shows that the compass is calibrated and Navigation is safe (green circle shown for both), and the "Hardware health" shows a red circle.  Clearly navigation is not safe if GPS and Mag are enabled but they are not hooked up to the board. 
## Steps to Reproduce

1.  Configure GPS and Magnetometer in Configuration Tab
2.  Disconnect (or do not connect in the first place) GPS and Magnetometer unitws
3.  Observe settings as shown in attached screenshot

## Expected behavior

The Compass calibrated and Navigation is Sage circles should be Red, not Green

## Suggested solution(s)

Enable the indicators to be correct
## Additional context

https://pastebin.com/FRF3uk6j
---

- FC Board name and vendor: Matek F722 SE
- INAV version string:
# version
# INAV/MATEKF722 2.4.0 Feb 11 2020 / 22:48:59 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

kelleysislander, 2020-04-19:
[image]

kelleysislander, 2020-04-19:
[image]

digitalentity, 2020-04-19:
This is working as intended. The names are for pre-flight checks.

"Navigation is safe" means that you don't have any navigation-aware flight modes configured, therefore there is no chance that your aircraft will need to navigate autonomously. Therefore the "Navigation safety" check passes.

"Compass calibrated" checks for the fact that calibration was done when the sensor is present. If sensor is not present - compass calibration values makes no sense and are not validated. Firmware makes no distinction between "sensor not connected" and "sensor not configured".

"Hardware health" sensor fails correctly and is enough to be blocking.

kelleysislander, 2020-04-19:
Thanks for that. 

Yes I do have Navi modes configured. 

Please take a look at the Aux section to see my modes. 

Cheers,

Bill

stronnag, 2020-04-19:
You do not appear to have GPS or compass modes defined:
```
ARM                  AUX3 1300 1700	(aux 0 0 3 1300 1700)
ANGLE                AUX0 1225 1950	(aux 1 1 0 1225 1950)
AIRMODE              AUX0 1125 1950	(aux 2 20 0 1125 1950)
NAVALTHOLD           AUX0 1750 1925	(aux 3 3 0 1750 1925)
BEEPERON             AUX3 1700 2100	(aux 4 11 3 1700 2100)
LEDLOW               AUX1 1300 2100	(aux 5 12 1 1300 2100)
OSD                  AUX1 1700 2100	(aux 6 15 1 1700 2100)
OSDALT1              AUX2 1075 1300	(aux 7 32 2 1075 1300)
OSDALT2              AUX2 1300 1700	(aux 8 33 2 1300 1700)
OSDALT3              AUX2 1700 2100	(aux 9 34 2 1700 2100)

```

kelleysislander, 2020-04-19:
Thank you for explaining, I get what you are saying now;)  If those were enabled the warnings would have responded differently.

digitalentity, 2020-04-20:
Yes, exactly. We could probably treat this as a feature request to report the irrelevant checks as "not checked".