iNavFlight/inav #5619: Run Time Calibration

HawaiiChad, 2020-04-25:
Hi all,

A little back story to my question/request. :)

Ever since getting into the hobby of RC I had the dream of being able to use a drone to spot and track schools of fish out on the open ocean, similar to how spotter planes track schools of big tuna.   Since early 2019 I have been testing different planes, cameras, and landing methods.  I have since then decided the following:

Airframe: Nano Talon (chosen because of its portability, space being limited on a boat)
Camera : DJI HD Fpv Goggles w/ Cyclops antenna.
Landing Method: Elastic Netting between two fiberglass posts set in rod holders.  

In January I planned to do my first sea trial and launch it from the boat a few miles off-shore.  Unfortunately Inav wouldn't arm.  I'm guessing that this is due to movements of the boat and Inav thinking one of the sensors are faulty.  Using speedybee its showing that pre-flight calibration failed.   Inav is the first and only flight software that I have used and I would like to continue using it to complete my project.  **Does anyone have a solution on how I can arm Inav while on a rocking boat?**

Much Mahalo!
Chad

Some photos below

First net landing.
[image]

Future net prototypes
[image]

[image]

Aerial view of a school of Halalu.
[image]

Some ocean views
[image]

wardnb, 2020-04-27:
can you see the arming warnings in the OSD or while connected to the inav-configurator? This could be the arming angle, which I believe is configurable: '''set small_angle 25''' 25 is the default for mini-quads I believe.

HawaiiChad, 2020-04-29:
> can you see the arming warnings in the OSD or while connected to the inav-configurator? This could be the arming angle, which I believe is configurable: `set small_angle = 25` 25 is the default for mini-quads I believe. You may need to increase this for your scenario if the waves are causing you to deviate from 25% angles.

Thank you for your reply!

From the last time I was out I do not recall it saying UAV is unlevel, but I hope that is the case. I changed small_angel = 50.  I'm going out tomorrow to test, will report back.

giacomo892, 2020-04-29:
It's not about small_angle, it's the gyro calibration process which needs stationary UAV to complete.
Ignoring that will led to poor flight performance. You might calibrate the plane on a gimbaled platform 👯

HawaiiChad, 2020-04-29:
> It's not about small_angle, it's the gyro calibration process which needs stationary UAV to complete.
> Ignoring that will led to poor flight performance. You might calibrate the plane on a gimbaled platform

That is what I was thinking.  Is there a way to calibrate on land and have it store the calibration? I was even considering plugging with my Speedy Bee usb c cable to my mobile phone, leaving it plugged in after departing on the boat, then plugging in the battery when ready to launch. Is there an easier way to do this?

giacomo892, 2020-04-29:
You can do that, indeed.
Not for the moment, since gyro offsets varies with temperature too

closed 2020-05-27 by DzikuVx