iNavFlight/inav #5621: Serial protocol for combined GPS+Baro+mag

spleenware, 2020-04-25:
## Current Behavior
Currently, we need to wire GPS to a UART, baro and mag to I2C. (I know NAZA gps can combine GPS+mag)

## Desired Behavior
To develop a protocol (like NAZA) which serialised GPS+baro+mag over the one UART.

## Suggested Solution
I'm not sure on the suggestion, but for there to be specific frames/sentences for baro readings, and specific frames for mag readings.

## Who does this impact? Who is this for?
For anyone, but mostly to have simpler wiring AND a broader FC board support. A lot of smaller AIO boards don't break out I2C so aren't candidates for INAV builds. Would be great to be able to make much smaller INAV quadcopter builds, which utilise these boards, but with a dedicated 'sensor' pod (eg. BN-180+ BMP280 + mag) which is away from interference.

## Additional context

digitalentity, 2020-04-25:
Makes sense. I'm working on something similar already, with some other extra functionality as well.

digitalentity, 2020-11-14:
Closed via #6090

closed 2020-11-14 by digitalentity