iNavFlight/inav #5636: Mixer settings for a singlecopter?

nickoe, 2020-04-27:
The Ardupilot does has a good overview of my setup, what I have is what [they call a singlecoplter ](https://ardupilot.org/copter/docs/singlecopter-and-coaxcopter.html) (or similar to a T-Hawk drone). 

That is; one propeller and four flaps.

Ardupilot has support for this but they don't support my flight controller. I would like use to use, and if needed add the support in inav if you can point me at the correct source files.

I think they mixing relations ships are quite similar to the simulation model in https://github.com/ArduPilot/ardupilot/blob/f7b47679bb0db22e7093d85c16a5ff9c791d87bd/libraries/SITL/SIM_SingleCopter.cpp#L62-L66

The flight controller I have is a [Mateksys F722-PX](http://www.mateksys.com/?portfolio=f722-px)  their PDB with servo outputs [F722-PX-W](http://www.mateksys.com/?portfolio=f722-px-w)

The problem I have is that in Inav Configurator, I can only add up to two motors and up to eight servos. But servos does not let me map moments for throttle, roll, pitch and yaw like we can for motors. So I have tried to add more motors, but still only two motors show up on the mixer mapping table. See the red boxed in the attached image. 
[image]

I have wired the S10 on the PDB to the serial for S.Bus between the flight controller and receiver. My propeller ESC is with PWM input. I would like to use S7 for the motor. My intention is to be able to control it with throttle, roll, pitch and yaw. 

I hope I can make it possible to use inav with singlecopters, but I think I need some input here. :)

DzikuVx, 2020-04-28:
Your platform will be "Multirotor", not "Other" and you will have to setup 1 motor and 4 servos with correct roll/yaw/pitch mixes assigned. You are on your own with weights on servos since those are very unique designs

closed 2020-04-28 by DzikuVx

nickoe, 2020-04-28:
@DzikuVx Thank you for pointing out that I need to use a multirotor to be able to assign more motors. But the next issue I see with this is; that the motors does not act as servos in that they don't take a mid value.

So when in static position on a table, I would expect them to be vertical or in the mid range, but they are rather at low at around 1150 instead of being at mid around 1500.

My motor mix looks like:
```
# dummies                                                                       
mmix 0 1 0 0 0                                                                  
mmix 1 1 0 0 0                                                                  
# flaps                                                                         
mmix 2 0.001 0 1 0                                                              
mmix 3 0.001 1 0 0                                                                                                                                                                                                                             
mmix 4 0.001 0 -1 0                                                             
mmix 5 0.001 -1 0 0                                                             
# motor                                                                         
mmix 6 1 0 0 0 
```

If I use servo mix for the servos, I found that I can use a negative weight to make the flap pair move correctly for roll and pitch, but the I don't have any control over yaw with the flaps.

If I then try to move around the physical connections I will be able to move the servos to S5 through S8, but it does not look like any platform mixer config will let me do that. It puts two servos on S9 and S10 instead. The motor is no on S4.

Is this last thing a bug?

DzikuVx, 2020-04-29:
The approach with `smix` is the correct one. You need one `mmix` with only throttle mixed in to it
and 8 `smix` rules for 4 servos.
Each servo should have mixed in STABILIZED YAW and either STABILIZED ROLL or STABILIZED PITCH.

Look here but remember that in this example `smix` command syntax is obsolete and you can not copy paste

nickoe, 2020-04-29:
@DzikuVx Ok, thank you. I will try that, with the eight servo rules. Is there a way to define a custom "Output Mapping" via the CLI? As mentioned previously all, I can make it do is have non-ideal motor and servo definitions to the numbers S outputs.

Did you forget a link in your last message?

nickoe, 2020-04-30:
@DzikuVx  Ok, it looks like I got it working, but I had to modify the firmware to be happy with my mounting of servos and the motor. See https://github.com/nickoe/inav/compare/e31c9f2b2a46f63183ab8be478c1534bcc12fd82...0d8cba46c189b96535702ea4ee32309f87d6c6f7

That is that I added a couple more `TIM_USE_MC_MOTOR` and moved them around such that I had the motor and servos on separate timers. I assume the `MC` part of that refers to MultiCopter and the `FW` equivalent is for FlyingWing or airplane in general.

And adding the extra servo mixer rules indeed seem to address my concern about controlling yaw with the servos as well as roll and pitch. This is my mmix and smix:
```
# mmix
mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000
mmix 2  1.000  0.000  0.000  0.000
mmix 3  1.000  0.000  0.000  0.000

# smix
smix 0 3 0 100 0 -1
smix 1 4 1 -100 0 -1
smix 2 5 0 -100 0 -1
smix 3 6 1 100 0 -1
smix 4 3 2 50 0 -1
smix 5 4 2 50 0 -1
smix 6 5 2 50 0 -1
smix 7 6 2 50 0 -1
```

Tinkerer1234, 2020-09-16:
Nickoe,

Please see : https://github.com/iNavFlight/inav/issues/6131

I have a similar issue and would like your advice on how to fix my issue, if you would be so kind.

nickoe, 2020-09-16:
@Tinkerer1234 I hacked the firmware as I showed in the previous comment. https://github.com/iNavFlight/inav/issues/5636#issuecomment-621809212

NCTEE, 2021-12-05:
@nickoe I have build up a Monocopter too (1 motor, 4 servos) with a MATEK F722-SE and GPS module. Changed firmare code according to the description above etc. I am struggeling to make it fly stabile. I tuned the PIDs up and down but didnt find a config yet that works, so I think it might be something different. Did you manage to make it fly stabile?

nickoe, 2021-12-05:
@NCTEE  I never really did a free flight with it as I broke the propeller and need a new one, but I was able to make it hold an attitude in pitch and yaw when tethered, even if I disturbed it. So I did not really have any proper free flight.