iNavFlight/inav #5640: magnetometer align INAV setup

kibinek, 2020-04-28:
I do not understand relationship between variables
**align_mag**
and 
**align_mag_roll
align_mag_pitch
align_mag_yaw**
because description at [https://github.com/iNavFlight/inav/wiki/GPS--and-Compass-setup](url) looks quite vague and misleading to me.
If a magnetomater is turned e.g. 270° and flipped, isn't the setup **align_mag = CW270FLIP** the same as **mag_align_pitch = 1800** & **mag_align_roll = 0** & **mag_align_yaw = 2700**?
Or in other words, which CLI setup of those is correct?
**align_mag = CW270FLIP
align_mag_roll = 0
align_mag_pitch = 0
align_mag_yaw = 0**
or
**align_mag = CW270FLIP
align_mag_roll = 0
align_mag_pitch = 1800
align_mag_yaw = 2700**
or
**align_mag = DEFAULT
align_mag_roll = 0
align_mag_pitch = 1800
align_mag_yaw = 2700**
This shall be streamlined in both description and INAV configurator I guess.

DzikuVx, 2020-04-28:
I suggest using 
```
align_mag = CW270FLIP
align_mag_roll = 0
align_mag_pitch = 0
align_mag_yaw = 0
```
since this is the only combination that can be set up from GUI

closed 2020-04-28 by DzikuVx

brainbubblersbest, 2020-11-15:
These are the Values that matches the Hardcoded Settings in the Configurator

This is out of the Source Code from Betaflight! but the Values are the same for INAV

#define CUSTOM_ALIGN_CW0_DEG ((sensorAlignment_t){ .values = { .roll = 0, .pitch = 0, .yaw = 0 }})
#define CUSTOM_ALIGN_CW90_DEG ((sensorAlignment_t){ .values = { .roll = 0, .pitch = 0, .yaw = 90 }})
#define CUSTOM_ALIGN_CW180_DEG ((sensorAlignment_t){ .values = { .roll = 0, .pitch = 0, .yaw = 180 }})
#define CUSTOM_ALIGN_CW270_DEG ((sensorAlignment_t){ .values = { .roll = 0, .pitch = 0, .yaw = 270 }})
#define CUSTOM_ALIGN_CW0_DEG_FLIP ((sensorAlignment_t){ .values = { .roll = 0, .pitch = 180, .yaw = 0 }})
#define CUSTOM_ALIGN_CW90_DEG_FLIP ((sensorAlignment_t){ .values = { .roll = 0, .pitch = 180, .yaw = 90 }})
#define CUSTOM_ALIGN_CW180_DEG_FLIP ((sensorAlignment_t){ .values = { .roll = 0, .pitch = 180, .yaw = 180 }})
#define CUSTOM_ALIGN_CW270_DEG_FLIP ((sensorAlignment_t){ .values = { .roll = 0, .pitch = 180, .yaw = 270 }})