iNavFlight/inav #5646: iNav 2.4.0 and Matek F411-WSE: no sensor data with smartport (sensor loss on OpenTX)

jocelyncloutier, 2020-04-28:
## Current Behavior

My set-up:
- Jumper T16 (latest OpenTx)
- R9M module (older pre-2019 model, latest firmware)
- Many R9 Slim+ receivers (older pre-2019 model, latest firmware)
- 4 x Matek F411-WSE (4 different FC) (All updated to iNav 2.4.0)

I am running into an issue where **I am losing (or have very spotty) sensor telemetry** coming from the FC over smartport. Please note that I am getting RSSI and RxBt telemetry from Slim+, so it is NOT a telemetry issue, but rather seemingly a sensor issue (data not getting from F411-WSE to Slim+).

Note about F411-WSE:
* Smartport is configured over softserial2 per Matek F411-WSE instructions.
** See attachment for iNav 2.4.0 diff all file [INAV_cli_Gob2_F411_20200426_110522.txt](https://github.com/iNavFlight/inav/files/4548880/INAV_cli_Gob2_F411_20200426_110522.txt)

IMPORTANT:
* Please note that there is NO problem with iNav 2.3.0
** I also attached the iNav 2.3.0 diff all for comparison purpose. 
[INAV_cli_Gob2_F411_20200405_065955.txt](https://github.com/iNavFlight/inav/files/4548883/INAV_cli_Gob2_F411_20200405_065955.txt)

## Steps to Reproduce

1. Configure F411-WSE to use smartport on softserial2
2. Discover sensor on T16
3. Usually within 2 minutes, the sensor telemetry will stop arriving on the T16.
** Looking at the OpenTx telemetry log, I can see some values showing up once in a while (like once ever 2 minutes)

Note: I was able to duplicate above behavior on 3 different Matek F411-WSE, with different Slim+ receiver and 2 different R9M module. Considering that this RSSI and RxBt telemetry is received on the T16, I do not believe that the issue with with the R9 system or openTx or T16. Finally, I do not have this problem when swapping the F411-WSE with a F722.

## Expected behavior
I am expecting to have the sensor telemetry consistently received on the T16.

## Suggested solution(s)
Unknown. I am reverting to iNav 2.3.0 as a workaround for now.

## Additional context

---

# version
# INAV/MATEKF411SE 2.4.0 Feb 11 2020 / 23:07:15 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

- FC Board name and vendor: Matek F4110WSE
- INAV version string: INAV/MATEKF411SE 2.4.0 Feb 11 2020 / 23:07:15 (bcd4caef9)

leeph, 2020-05-11:
I would recommend you switch to F.port.

Softserial is not as reliable as a hardware UART. Slim+ has an inverted smartport output so you only need one signal conductor to F411-WSE TX pin for both control and telemetry. I have many craft setup like this and telemetry is wholly reliable.

jocelyncloutier, 2020-05-13:
Yes, this issue pushed me to look for a work around and I did install the FCC FPORT firmware to all of my Slmi+ receiver and configured by FC accordingly... it is a much more elegant solution, with RSSI in channel 16 as a bonus.

thanks!

mrcottonmouth, 2020-06-19:
I was getting the same problem, with a Frsky R9. intermittent telemetry. I swapped out the R9 receiver, no luck. So I upgraded to a TBS Crossfire. That changed to me using tx and rx on UART2 with Crossfire's protocol (instead of Sbus and TX). I still have the problem. It drives me crazy trying to fly with the LUA script constantly repeating itself and saying incorrect things. I am running INAV/MATEKF411SE 2.4.0.  I have an almost exact setup in another plane with no issues. Is it a bad board? The only thing I can think to try, is to reflash the f411se, but I really don't think that will help either.

mrcottonmouth, 2020-06-20:
I upgraded to 2.5.1 and the issue seems resolved.

FPWibi, 2020-06-24:
Are you sure it's resolved? 
I'm using an R9MM with F.Port Firmware on the Matek F411-WSE.
After 3-5 Minutes all Sensors get lost. RSSI and RX-Bat are still there. Control input also keeps working. 
I tested this Receiver on two other FCs and it works with them.
Updating to Inav 2.5.1 didn't solve the issue for me.
I'm sure it's an issue with the FC. Maybe because softserial and uart is on the same pad.

leeph, 2020-06-24:
Issue reported is for R9MM using Smartport and SBUS.

You are running F.port and so there is no separate Smartport wire, nor should you be using softserial.

I have a number of R9MM running on F411-WSE, all running F.port, and I have no issue with telemetry dropping out on any version of iNav including latest 2.5.1.

FPWibi, 2020-06-25:
I am using only one wire from the inverted s. Port to the TX2 Pad on the FC.
The description of the FC says that uart2 and softserial2 share this TX2 pad. 
I tried using this pad as softserial with Sport und as uart with F.Port. In both cases I had problems with telemetry dropping out. But it is not a problem with the receiver, because the receiver works with other FCs and RSSI and RX-Bat do not drop out. 

Do you have your F.Port connected to the TX2 pad?

closed 2020-09-07 by stale[bot]

ehubin, 2021-01-26:
I have similar issue on Inav 2.6.1 I am using R9 SX which has no inverted hack as far as i know so i am stuck and cannot use the FPORT workaround.

Any chance this issue could be looked at?

Viper1200, 2021-04-16:
https://github.com/iNavFlight/inav/issues/6836#issue-858577832