iNavFlight/inav #5648: SBUS Output

robthomson, 2020-04-29:
Would it be possible to enable as PWM channel - or a serial channel to be configured to output sbus or ppm?

Sbus would be good. This would enable a simple solution to drive multiple sbus servos from the flight controller.

Functionally.. on bigger models this can simply things as you can then use remote sbus to pwm convertors out on the wings - net result.. less wiring!

digitalentity, 2020-04-29:
Should be doable.

robthomson, 2020-04-29:
Fantastic.  Would be really helpful.   I dare say easier for many than then i2c servo expander!

digitalentity, 2020-04-30:
Actually, this is a very low hanging fruit. I'll look into it for 2.5

robthomson, 2020-04-30:
Great stuff.  No major rush my end.. But will really simplify future builds.

Goodnto know its in the pipe line.

digitalentity, 2020-04-30:
@robthomson I might have a draft ready by weekend, but I don't have any SBUS servos at hand. Do you have a setup you can test with?

robthomson, 2020-04-30:
Yup.  I can do this easily.

Another simple option... 

Feed the sbus into another flight controller that can receive sbus. 

But more than happy to test this end if you want. want

helicocrasher, 2020-05-01:
Fantastic, interrested in this function & could do some testing with discrete sbus decoders

advancetom, 2020-05-05:
this would be also useful for use as a helicopter flybarless mode!!

robthomson, 2020-05-05:
Hoping to test this again tomorrow.. been a bit snowed under past couple of days!

closed 2020-05-08 by digitalentity

Enggmaug, 2021-05-06:
Hello, I struggle finding how to know/define where this SBUS output is mapped to ?
I suppose it won't be the same as the SBUS input from the receiver... or will it ?

advancetom, 2021-05-06:
Follow

moyouyilansefengye, 2021-05-16:
I am a little confused about outputting sbus. I am running Inav 3.0.0 on my F4V3 and configurator is also 3.0.0.
I want to use an FC to convert crsf signal to sbus. Is that possible?

VancouverUmbrella, 2021-07-11:
Pass-through SBUS output from a FC is just what I'm looking for. 

It's not clear to me whether the request for SBUS output in iNAV has been implemented or not.

Can anyone set me straight and point me to documentation?

Many thanks.

Enggmaug, 2021-07-11:
I put the following :
```
serial 2 4194304 115200 115200 0 115200
set servo_protocol = SBUS
```

And it jsut worked outputing SBUS on Serial 3. 
Need then a Serial to SBUS inverter, and SBUS to Servos converter.

G.

VancouverUmbrella, 2021-07-11:
Wonderful!

Thank you for that. Much appreciated.

Kindly,
Andrew

froqstar, 2022-07-17:
@Enggmaug do you happen to know where to find documentation on that feature, e.g. how you came up with that CLI command? I struggle to find anything...

Enggmaug, 2022-07-20:
> 

Hey, I don't quite remember, I believe it came from some forum discussions I found in google.
In the end, I'm not entirely satisfied, my setup wasn't very reliable. But at this point, I'm not sure whether it is due to the Inav implementation, the FC, the Sbus to PWM converter, or overall wiring and power distribution.
What happens is that sometimes, all servos take a random position and won't move anymore... it is possibly a power issue, since I use 7 servos on a biplan... 
I gave this up. I'd very much like to try out a similar solution using crossfire, though. I have a mateksys CSRF to PWM converter that allows for a better power distribution. I already made the feature request. Not sure it will be implemented, though...

G.