iNavFlight/inav #5649: Curves

robthomson, 2020-04-29:
Currently inav outputs a rather linear response to a servo.

Nothing wrong with this - but the ability to apply a curve to a 'mix' would be fantastic,.

This would be very helpfull on more complex models where you need to consider curves as integral to good flight performance. (e.g.  FPV soaring)

digitalentity, 2020-04-29:
PID works best with linear systems. Adding a curve to linearise response of the aircraft vs servo command would indeed be beneficial. We need some math to back that though.

robthomson, 2020-04-29:
I imagine I might be able to get some help from the opentx guys if that would help?

digitalentity, 2020-04-30:
Adding a random curve on a servo output is easy. Adding a curve that has a physical meaning - this is more complicated. Before we go in and start adding stuff, it would be awesome if somebody will provide data to help figure out shape of the curve.

robthomson, 2020-04-30:
I would see this being a case of having a 'curves' section on the configurator.  You can then create X number of curves.

Then.. on the mixer page.. you can associate a curve to the mix.

Curves can have many forms.  

- multi point
- x/y movable
- 3 point, 8 point etc...

robthomson, 2020-04-30:
All info to do this should be in opentx project in:  radio / src / curves.cpp

Certainly the maths is available :-)

digitalentity, 2020-04-30:
@DzikuVx WDYT? Time to look into custom curves in 2.6?

DzikuVx, 2020-04-30:
That sounds like an idea. Can not promise anythingh since I want to put a few more bits a pieces for VTOL

jbishop129, 2020-05-13:
Hi @DzikuVx  - you mention bits and pieces for VTOL in an upcoming release; I’d love to know more about this! I’m building a ‘quadPlane’ VTOL (based on the FliteTest Bronco airframe, built my own tilt mechanisms for front props, and two two fixed-position rear props for the ‘quad’ mode). Your comment makes me optimistic there’s some bits coming that are of interest for VTOL builders?!

DzikuVx, 2020-05-14:
@jbishop129 there are bits but nothing that was tested or ready to be used

closed 2022-05-07 by DzikuVx