iNavFlight/inav #5650: Lost Frame Rate in OSD

namirda, 2020-04-29:
## Current Behavior
Latest FrSky ACCESS firmware for R9 hardware provides a new telemetry item called Lost Frame Rate which I understand to be the number of lost or corrupted telemetry frames per second. LFR can be considered as something analogous to the LQ of Crossfire providing additional information about the integrity of the telemetry link. It is not currently possible to display LFR in the INav OSD unless you assign it to the RSSI channel in OpenTx.

## Desired Behavior
Provide native support for LFR in the INav OSD

## Who does this impact? Who is this for?
All users of FrSky R9 hardware.

## Additional context
There is native support for LQ in Betaflight since version 4.1 - something similar for LFR and LQ in INav would be great.

digitalentity, 2020-04-30:
I understand this is made available on the radio. Unless FrSky exposes this to the FC via S.Bus or other radio control protocol, it will be impossible to show on OSD, except by doing a loop-back via a, RC channel.

namirda, 2020-04-30:
Yes LFR is available on the radio and it is easy to set an alarm in OpenTx. 
I assumed it was also available to the FC via FPort in the same way that RSSI is. 
How can I check this?

digitalentity, 2020-05-01:
F.Port is essentially S.Bus + S.Port via the same wire. S.Bus doesn't have a field for RSSI or LQ, S.Port is only used to poll telemetry values **from** the FC. So, I don't think this is possible, unless FrSky created a new F.Port frame to send these values to the FC.

namirda, 2020-05-01:
Thanks DE. 

I have just noticed that a similar issue was raised recently on the betaflight github https://github.com/betaflight/betaflight/issues/9680 and the guys there are waiting for some response from FrSky. 

I will close this for now and reopen if/when there is some progress made over there.

digitalentity, 2020-05-02:
Let's keep if open so we don't forget

VitaliyNimych, 2020-07-16:
https://github.com/VitaliyNimych/betaflight/commit/ea9ab071d854a65f5a8b388ffc6fc69e4b346cb0

VitaliyNimych, 2020-07-16:
https://github.com/betaflight/betaflight/pull/10019

VitaliyNimych, 2020-07-16:
https://www.youtube.com/watch?v=d2Yf21Ct6PY&feature=youtu.be

VitaliyNimych, 2020-07-16:
https://github.com/opentx/opentx/pull/7661/commits/fea8d6952052ff70513878d6078332c69343d27a

DzikuVx, 2020-11-30:
seems like the idea died on BF side as well

closed 2020-11-30 by DzikuVx