iNavFlight/inav #5657: Hi as there are some sensors only show as an ID instead of a name, so where I can find a list of all the sensors with all the name and unit etc. for example.

xixijoe, 2020-05-01:
For immediate help, just ask your question on one of the following platforms:

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)

You can also read public documentations or watch video tutorials:

* [Official documentation](https://github.com/iNavFlight/inav/tree/master/docs)
* [Official Wiki](https://github.com/iNavFlight/inav/wiki)
* [Video series by Painless360](https://www.youtube.com/playlist?list=PLYsWjANuAm4qdXEGFSeUhOZ10-H8YTSnH)
* [Video series by Paweł Spychalski](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLloACrA6f1_daCjhqY2x0fB)
* [How to setup INAV on a flying wing](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLn_lCX7UB2OPLzR4w5G4el3)
* [How to setup INAV on a 5" quad](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLl9M6Vm-ZNLMCr6_9yNVHQG)

DzikuVx, 2020-05-01:
Please refer to OpenTX. Sensors not officially supprted have no name.
And by the way, you've been reminded several times not to use GitHub issues as a way of support and stop asking questions in topic.

closed 2020-05-01 by DzikuVx

xixijoe, 2020-05-01:
> Please refer to OpenTX. Sensors not officially supprted have no name.
> And by the way, you've been reminded several times not to use GitHub issues as a way of support and stop asking questions in topic.

sorry, i did not know that i have been reminded for such a rule. can you tell what did I do wrong here? thank you