iNavFlight/inav #5660: How to set Hover throttle when "set nav_use_midthr_for_althold = ON

Ravid824, 2020-05-02:
In bug #4182 @shellixyz  explained nicely the relation between the parameters and how to set the Hover throttle value properly. 
All was clear until latest releases (I am on 2.4) when min_throttle was changed to throttle_idle. In release 2.4 still min_throttle still exist and shows
cli? get min_throttle
fw_min_throttle_down_pitch = 0 !!!!

Can you please explain the relations between the new parameters and how to set the value nav_mc_hover_thr properly. (example will be helpful).
set nav_use_midthr_for_althold = OFF works OK but it is easier and safer to use it when it is ON.
Thank you

evildronepilot, 2020-05-09:
why wouldn't you take current throttle position and assign it to a a hover point position  dynamically instead

digitalentity, 2020-05-09:
We can't assume that current throttle is hover. Althold may kick in at zero throttle or at full throttle. `nav_mc_hover_thr` is hint to the firmware what the hover throttle is.

`nav_use_midthr_for_althold` doesn't control which throttle will go to motoros. It controls only where the user perception of hover is (current stick position or stick middle)

Ravid824, 2020-05-09:
So what throttle value I should enter. Let say that my drone hover at throttle 1560. it is OK to set this number and assume that if the throttle around this value it will keep ALTHold?
I don't think that this is the current behavior.

Mateyhv, 2020-06-04:
> So what throttle value I should enter. Let say that my drone hover at throttle 1560. it is OK to set this number and assume that if the throttle around this value it will keep ALTHold?
> I don't think that this is the current behavior.

I think you are mixing things. Tune nav_mc_hover_thr just as you said: hover, take note of the ms value of throttle channel and set it.

set nav_use_midthr_for_althold will make your throttle stick to act as up and down command for ALTHOLD. When you switch to ALTHOLD If ON it will take middle stick as neutral. If OFF current throttle stick position will be neutral. The transition will be smoother but the stick travel will be unequal up and down from the new "neutral" position and thus the max up and down speed will be unequal either