iNavFlight/inav #5661: [REQUEST] Autolaunch time delay before nav_fw_launch_idle_thr	kicks in

b14ckyy, 2020-05-02:
## Current Behavior
When Autolaunch is used with _nav_fw_launch_idle_thr_ set the motor will spin up as soon as the throttle stick is raised.

## Desired Behavior
There should be a possibility to set a delay time between raising the throttle stick and motor starts spinning by _nav_fw_launch_idle_thr_
After my video about why shaking before launch is a bad idea (https://www.youtube.com/watch?v=3HULgXW6VXY) some people explained that they have very big and heavy setups that must be held with 2 hands all the time before the throw and they indeed need a spinning motor when they are thrown. But a high idle throttle will be dangerous as they need to move one hand off the plane to raise the throttle if the motor spins up directly. That's why they still shake it to launch the motor manually, what can cause unstable launches due to low thrust to weight ratio. 

## Suggested Solution
It would be handy to have a additional timer that can add a delay between throttle stick up and idle throttle starting the motor. Let's say the delay is set to 10s. The Pilot arms the plane, raises the throttle, then he has enough time to get ready and hold the plane tight. The countdown ends and the motor gradually spins up to idle throttle. Then the pilot throws the plane and autolaunch kicks in and continues the launch sequence by it's own. 

Maybe a buzzer alarm when the timer has started would add extra security so the pilot can get ready. 

## Who does this impact? Who is this for?
All pilots with big and heavy planes launching them by hand and not with a bungee and want to use Autolaunch in the correct way. 

## Additional context
For example a friend has a ZCon that weights about 4.5kg. It must have the motor spinning at 60-70% thrust before it is thrown and it makes sense to use idle throttle for this. But then it has about 3kg of static thrust and it is very dangerous to hold it in just one hand while the other one is still on the throttle stick.

b14ckyy, 2020-07-04:
nope :)

b14ckyy, 2020-09-02:
nope

MrD-RC, 2020-11-07:
This is a good idea. Maybe also a spin up time too. So that if the idle throttle is something like 70%, as for that Zcon, it builds up to that over a second or so, to lessen the chance of losing grip of the model.

b14ckyy, 2020-11-07:
Idle throttle has a hardcoded spin up time of 2s already. that's fine. Just a delay would be fine.

b14ckyy, 2021-05-10:
thx @jaroszmm for the reminder. Totally forgot about that already. :D

MrD-RC, 2021-05-10:
https://github.com/iNavFlight/inav/pull/6943

breadoven, 2021-11-29:
Closed by #6943 I think.

closed 2021-11-29 by breadoven