iNavFlight/inav #5662: possible bug or is it intended behaviour? No osd until RX is powered

wx4cb, 2020-05-02:
Matek f405/f722/f765 wing, crossfire receiver

Just a quick question to clear something up. whenever i power up the flight controller (outside of usb) i don't get any osd (splash screen or otherwise) until the receiver is turned on.

For example, if i leave the transmitter off and power up the plane to start getting GPS data there is no OSD. as soon as the transmitter is turned on, the normal osd appears.

is this a bug or is this correct ?

giacomo892, 2020-05-03:
Post your diff

wx4cb, 2020-05-03:
> Post your diff

i thought i had. be here it is

```
# diff

# version
# INAV/MATEKF722SE 2.4.0 Feb 11 2020 / 22:53:38 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# start the command batch
batch start

# resources

# mixer
mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix 0 3 0 -50 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 50 0 -1
smix 3 4 1 50 0 -1
smix 4 5 16 100 0 -1

# servo
servo 3 1000 2000 1499 100
servo 4 1000 2000 1671 -100

# logic

# gf

# feature
feature -TX_PROF_SEL
feature MOTOR_STOP
feature DYNAMIC_FILTERS
feature GPS
feature LED_STRIP
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial
serial 0 0 115200 115200 115200 115200
serial 1 2048 115200 57600 0 115200
serial 2 2 115200 115200 0 115200
serial 3 256 57600 115200 57600 115200
serial 5 64 115200 115200 0 115200

# led
led 0 6,1:SW:CTB:2
led 1 7,1:S:CTB:1
led 2 8,1:ES:CTB:2
led 3 7,3:N:CTB:1

# color

# mode_color

# aux
aux 0 0 0 1925 2100
aux 1 1 1 1125 1275
aux 2 3 1 1350 1450
aux 3 3 1 1550 1650
aux 4 9 1 1550 1650
aux 5 8 1 1925 2075
aux 6 19 0 1725 1900
aux 7 35 1 1350 1450
aux 8 10 1 900 1075
aux 9 27 2 1925 2100
aux 10 28 3 1925 2100
aux 11 32 4 1450 1550
aux 12 33 4 1925 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 26 0 V
osd_layout 0 1 1 0 V
osd_layout 0 2 0 0 V
osd_layout 0 3 8 6 H
osd_layout 0 4 8 6 H
osd_layout 0 7 17 14 V
osd_layout 0 9 23 1 H
osd_layout 0 11 7 0 V
osd_layout 0 12 1 1 V
osd_layout 0 13 19 1 V
osd_layout 0 14 6 14 V
osd_layout 0 15 7 1 V
osd_layout 0 20 1 14 H
osd_layout 0 21 1 13 H
osd_layout 0 22 15 14 V
osd_layout 0 23 19 0 V
osd_layout 0 24 13 1 V
osd_layout 0 26 12 11 H
osd_layout 0 28 23 14 V
osd_layout 0 29 23 10 H
osd_layout 0 30 1 4 V
osd_layout 0 31 1 14 V
osd_layout 0 33 24 1 V
osd_layout 0 34 11 1 H
osd_layout 0 40 12 0 V
osd_layout 0 41 25 12 H
osd_layout 0 42 21 12 H
osd_layout 0 45 0 0 V
osd_layout 0 46 23 14 H
osd_layout 0 47 16 14 H
osd_layout 0 50 7 14 H
osd_layout 0 85 8 14 H
osd_layout 0 86 23 13 H
osd_layout 0 87 1 13 H
osd_layout 0 88 2 7 H
osd_layout 0 89 14 1 H
osd_layout 0 97 1 13 V
osd_layout 0 98 23 13 V
osd_layout 1 0 26 0 V
osd_layout 1 1 1 0 V
osd_layout 1 7 17 14 V
osd_layout 1 9 23 1 H
osd_layout 1 11 7 0 V
osd_layout 1 12 1 1 V
osd_layout 1 13 19 1 V
osd_layout 1 14 6 14 H
osd_layout 1 15 7 1 V
osd_layout 1 20 1 14 H
osd_layout 1 21 1 13 H
osd_layout 1 22 14 14 V
osd_layout 1 23 19 0 V
osd_layout 1 24 13 1 V
osd_layout 1 26 12 11 H
osd_layout 1 28 23 14 H
osd_layout 1 29 23 10 H
osd_layout 1 30 1 4 V
osd_layout 1 31 1 14 H
osd_layout 1 33 24 1 V
osd_layout 1 34 11 1 H
osd_layout 1 40 12 0 V
osd_layout 1 41 25 12 H
osd_layout 1 42 21 12 H
osd_layout 1 45 0 0 V
osd_layout 1 46 23 14 H
osd_layout 1 47 16 14 H
osd_layout 1 50 7 14 H
osd_layout 1 85 8 14 H
osd_layout 1 86 23 13 H
osd_layout 1 87 1 13 H
osd_layout 1 88 2 7 H
osd_layout 1 89 14 1 H
osd_layout 1 97 1 14 V
osd_layout 1 98 24 14 V
osd_layout 2 0 26 0 V
osd_layout 2 1 1 0 V
osd_layout 2 7 17 14 H
osd_layout 2 9 23 1 H
osd_layout 2 11 7 0 V
osd_layout 2 12 12 0 V
osd_layout 2 13 25 1 V
osd_layout 2 14 6 14 H
osd_layout 2 15 21 1 V
osd_layout 2 20 1 14 H
osd_layout 2 21 1 13 H
osd_layout 2 22 15 1 V
osd_layout 2 23 19 0 H
osd_layout 2 24 13 1 H
osd_layout 2 26 12 11 H
osd_layout 2 28 23 14 H
osd_layout 2 29 23 10 H
osd_layout 2 30 1 4 V
osd_layout 2 31 1 14 H
osd_layout 2 33 24 1 H
osd_layout 2 34 11 1 H
osd_layout 2 40 12 0 H
osd_layout 2 41 25 12 H
osd_layout 2 42 21 12 H
osd_layout 2 46 23 14 H
osd_layout 2 47 16 14 H
osd_layout 2 50 7 14 H
osd_layout 2 85 8 14 H
osd_layout 2 86 23 13 H
osd_layout 2 87 1 13 H
osd_layout 2 88 2 7 H
osd_layout 2 89 14 1 H
osd_layout 2 97 1 1 V
osd_layout 2 98 23 13 H

# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set acc_hardware = MPU6000
set acczero_x = 61
set acczero_y = -33
set acczero_z = -387
set accgain_x = 4098
set accgain_y = 4082
set accgain_z = 4021
set mag_hardware = NONE
set mag_declination = -600
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 12
set serialrx_provider = CRSF
set max_throttle = 2000
set motor_pwm_rate = 2000
set motor_pwm_protocol = MULTISHOT
set failsafe_procedure = RTH
set current_meter_scale = 250
set platform_type = AIRPLANE
set model_preview_type = 8
set servo_pwm_rate = 60
set small_angle = 180
set applied_defaults = 1
set gps_provider = UBLOX7
set nav_rth_climb_first = OFF
set nav_rth_allow_landing = FS_ONLY
set nav_rth_altitude = 6096
set nav_fw_cruise_thr = 1450
set nav_fw_max_thr = 2000
set nav_fw_climb_angle = 25
set nav_fw_loiter_radius = 3000
set osd_video_system = PAL
set osd_units = IMPERIAL
set osd_rssi_alarm = 0
set osd_time_alarm = 0
set osd_alt_alarm = 0
set osd_dist_alarm = 0
set osd_neg_alt_alarm = 0
set osd_crosshairs_style = TYPE7
set osd_left_sidebar_scroll = GROUND_SPEED
set osd_right_sidebar_scroll = ALTITUDE
set osd_coordinate_digits = 8
set osd_plus_code_digits = 10
set name = WX4CB-Harrier
set tz_offset = -240
set vtx_band = 5
set vtx_channel = 4
set vtx_power = 5
set vtx_freq = 5769

# profile
profile 1

set fw_ff_pitch = 70
set fw_ff_roll = 70
set fw_ff_yaw = 70
set fw_p_level = 40
set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 600
set rc_expo = 0
set rc_yaw_expo = 0
set roll_rate = 60
set pitch_rate = 60
set yaw_rate = 60
set manual_rc_expo = 0
set manual_rc_yaw_expo = 0

# battery_profile
battery_profile 1

# end the command batch
batch end
```

giacomo892, 2020-05-03:
You have OSD alternate layers misconfigured

wx4cb, 2020-05-03:
i figured it was something i did, but i thought that if i had that channel on 1500 which it should be without an rx it should go to one of the alternate osds?

giacomo892, 2020-05-03:
if you need more support, reach us on social media

closed 2020-05-03 by giacomo892