iNavFlight/inav #5663: DSHOT and telemetry issue related to blackbox logging rate

TcheTcheFPV, 2020-05-02:
## Current Behavior

The quad is disarming on its own and telemetry freeze.

When using DSHOT: The quad disarm few second after arming, and telemetry values (voltage, etc...) that are transmitted to the R/C (taranis x9d+), freezes (but still ok in the OSD). 

when using MULTISHOT : the quad does not disarm, but telemetry values (voltage, etc...) that are transmitted to the R/C (taranis x9d+), freezes few seconds after arming. (but still ok in the OSD).

Both case above can be reproduced each time, as long as blackbox logging rate is at 100%.
If blackbox logging rate is under 100% (ie 88%), then both issue with dshot and telemetry does not happen. Tried with 2 different SDcard and one is a high speed one.

So, it seems that the DSHOT issue and telemetry freeze is related to blackbox logging rate, and even if it's not happen when logging rate is under 100%, we should ensure it's ok.

## Steps to Reproduce

1. configure quad with dshot
2. configure blackbox logging rate at 100%
3. arm the quad
4. wait few second(5 to 30, should depend)

## Expected behavior

## Suggested solution(s)

## Additional context

---

# version
# INAV/KAKUTEF7 2.4.0 Feb 11 2020 / 21:39:46 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]
- FC Board name and vendor:
Kakute F7 v1.5
- INAV version string:
CONFIGURATOR 2.4.1

digitalentity, 2020-05-02:
Please provide a config  diff

TcheTcheFPV, 2020-05-02:
Hello, here it is :

diff

# version
# INAV/KAKUTEF7 2.4.0 Feb 11 2020 / 21:39:46 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# start the command batch
batch start

# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix 0 0 0 100 0 -1

# servo

# logic

# gf

# feature
feature -TX_PROF_SEL
feature BAT_PROF_AUTOSWITCH
feature DYNAMIC_FILTERS
feature PWM_OUTPUT_ENABLE

# beeper
beeper -RUNTIME_CALIBRATION
beeper -DISARMING
beeper -ARMING
beeper -BAT_LOW
beeper -ACTION_SUCCESS
beeper -ACTION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -LAUNCH_MODE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

# map
map TAER

# serial
serial 0 0 115200 115200 0 115200
serial 3 2048 115200 115200 0 115200
serial 6 262144 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1900 2100
aux 1 1 3 1300 1700
aux 2 3 3 1925 2100
aux 3 11 2 1300 1700

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 26 1 V
osd_layout 0 1 1 15 H
osd_layout 0 6 23 14 V
osd_layout 0 7 13 2 V
osd_layout 0 8 11 1 V
osd_layout 0 9 1 2 H
osd_layout 0 10 1 1 V
osd_layout 0 11 7 14 V
osd_layout 0 12 11 14 V
osd_layout 0 15 3 5 V
osd_layout 0 28 23 11 H
osd_layout 0 30 1 12 V
osd_layout 0 32 1 14 V
osd_layout 0 38 17 14 V

# master
set looptime = 250
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 90
set gyro_lpf_type = PT1
set gyro_stage2_lowpass_hz = 175
set acc_hardware = MPU6000
set acczero_x = 35
set acczero_y = 8
set acczero_z = -21
set accgain_x = 4089
set accgain_y = 4073
set accgain_z = 4100
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set serialrx_provider = CRSF
set max_throttle = 2000
set motor_pwm_rate = 16000
set motor_pwm_protocol = DSHOT600
set throttle_idle =  4.000
set current_meter_scale = 210
set yaw_motor_direction = -1
set model_preview_type = 3
set applied_defaults = 1
set mc_airmode_type = THROTTLE_THRESHOLD
set inav_w_z_baro_p =  0.350
set osd_video_system = PAL
set osd_crosshairs_style = TYPE3
set osd_left_sidebar_scroll = ALTITUDE
set osd_right_sidebar_scroll = GROUND_SPEED
set osd_sidebar_scroll_arrows = ON
set name = tchetche
set vtx_band = 5
set vtx_low_power_disarm = ON
set vtx_freq = 5658

# profile
profile 1

set mc_p_pitch = 45
set mc_i_pitch = 35
set mc_d_pitch = 24
set mc_i_roll = 35
set mc_d_roll = 24
set dterm_lpf_hz = 75
set d_boost_factor =  1.500
set d_boost_max_at_acceleration =  5000.000
set d_boost_gyro_delta_lpf_hz = 50
set rc_yaw_expo = 70
set roll_rate = 60
set pitch_rate = 60
set yaw_rate = 50

# battery_profile
battery_profile 1

set battery_capacity = 1550

# end the command batch
batch end

#

digitalentity, 2020-05-06:
Looks like blackbox can't cope with the 4kHz logging rate. We probably need to look into making  this more stable though.

DzikuVx, 2020-09-16:
See #6127 INAV 2.6 will allow faster logging rate than 1kHz

closed 2020-09-16 by DzikuVx