iNavFlight/inav #5670: ADDING I2C BAROMETER

rgp1942, 2020-05-03:
In an attempt to get altitude hold on a customized 110mm "cinewhoop" I have added a brand new Adafruit BMP388 [https://www.adafruit.com/product/3966] to a brand new Matek F405 Mini FC [http://www.mateksys.com/?portfolio=f405-mini] via the I2C connections.  The barometer is not being recognized by INAV 2.4.1 using the Matek firmware for the F405 -- 2.4.0 Feb 11 2020.

I have spent MANY man hours researching this on the github documention using searches like site:github.com/iNavFlight/inav/wiki BARO and site:github.com/iNavFlight/inav/wiki I2C.  Also, many Google searches related to the subject.

**What I am looking for are working instructions for adding an I2C barometer to INAV.**

I have triple checked the wiring, set inav>configuration>sensors to AUTO and BMP388, etc.  I can provide pictures, wiring diagrams, inav lists, or whatever might be helpful in getting this to work.  I suspect there are CLI commands I have failed to issue, but can find no "how to" anywhere on the internet for this situation.  Whatever you can point me to would be greatly appreciated.

THANKS!

digitalentity, 2020-05-04:
The real issue is that BMP388 is not supported on that board, we'll fix that

rgp1942, 2020-05-04:
Aha!  I'm on the "bleeding edge" again. I have an FC with a bmp280 in it where the baro just shows up, but I wasn't sure if the I2C attached required special CLI enabling or not.  I went with the bmp388 because the resolution is twice that of the 280,

Please let me know how I can be of assistance.  I am happy to beta test or provide whatever I can.  I can send documentation, or you can obtain it from the links above.

What do you anticipate will be the ETA of this change?  Thanks!

[image]

rgp1942, 2020-05-04:
[https://learn.adafruit.com/adafruit-bmp388](url)

rgp1942, 2020-05-04:
[http://www.mateksys.com/?portfolio=f405-mini](url)

rgp1942, 2020-05-04:
Would it work if I switched over to SPI bus?  The Adafruit BMP388 supports it.

digitalentity, 2020-05-05:
We may add support for BMP388 to boards without internal barometer in 2.5

rgp1942, 2020-05-05:
Sounds like bmp388/i2c support is a ways off.  In the meantime I've ordered a Matek 411-mini-se with bmp280 in it.

rgp1942, 2020-05-05:
What mini/20x20 FC's **DO** support an external BMP388 in INAV (either I2C or SPI)?

digitalentity, 2020-05-06:
@rgp1942 none at the time. Which board do you think should get support for BMP388?

rgp1942, 2020-05-06:
**Given:**
+ The BMP388 precision barometer has been available for over 2 years, and was specifically designed for drones.  It provides twice the accuracy of the previous BMP280, which is built into many flight controllers.  This new barometer has the potential to improve altitude hold for drones.
+ Adafruit makes a very nice BMP388 board that can be connected to an FC via I2C or SPI.
+ SP Racing appears to be the only manufacturer so far to incorporate the BMP388 into their FC's.
+ INAV is the only FC configurator that supports altitude hold (I think).
+ Once the BMP388 is "discovered" by FC makers, they will incorporate them.
**Therefore:**
+ INAV needs to at least support the SP Racing boards with BMP388's in them.
+ INAV needs to support the most popular INAV FC's that can have barometers attached via I2C/SPI.

In my particular case I'm trying to use a Matek F405-MINI with Adafruit BMP388 connected via I2C, so naturally I would like it to work yesterday!  :o)  In the meantime I've fallen back to ordering a Matek F411-MINI-SE with built-in BMP280.  Hopefully I will then have altitude hold in my 110mm Cinewhoop, but not as accurate as it could be.

I am willing to be a "test case" as needed.

THANKS!

digitalentity, 2020-05-06:
Given that there are no widely available FCs which are both INAV-supported and have BMP388 onboard, it makes little sense to chase this into 2.5 milestone which is right around the corner.

Support for H7 boards, including one from SPRacing is coming in 2.6.

rgp1942, 2020-05-06:
Would creating a "patch" for me to use with my F405+388 take the same effort as incorporating it into a future release?  If there's something I can do to jury-rig it I'm willing to give it a try.  Maybe a work-around using CLI commands?  Otherwise it sounds like only the 280 will be supported in INAV for some time to come.  In that altitude hold was removed from cleanflight and betaflight, it only leaves INAV for us more camera oriented non-racers.  I am a retired computer electronics and programming person, but don't want to get into to deep, but could probably cope with a patch.  Does the baro have some kind of ID that needs to be recognized?  I could tell INAV it's a 280, but INAV is not even recognizing a baro is there, so must be more to it.  Thanks for your efforts and communications so far.  **I really appreciate it!**  :o)  p.s. Bosch has now come out with a BMP390 for mobile devices.

digitalentity, 2020-05-07:
I can build you a firmware version that will support bmp388 on MATEKF405 target no problem, thanks for offering to test it.

rgp1942, 2020-05-07:
Thanks for the offer to make me a custom target.  I'm happy to test it.  Be advised the FC is a Matek F405-MINI (20x20), not the F405 (30x30), if that makes a difference.  giacomo892 said to use the F405 target.
https://github.com/iNavFlight/inav/issues/5639

MATEKSYS, 2020-05-08:
We compared MS5611 with BMP388 and BMP280 on different I2C2 bus of one Flight controller. identical power source. Seems BM388 doesn't preform well as what the manual said
[image]

salukikev, 2021-08-21:
So this is more than a year old now, but in that time, we now have BMP390, the most accurate barometer yet.  Looking forward on this, can we go ahead and implement support for that hardware?   A high accuracy external barometer is essential to my build, and I'm sure this will only be more popular in the future.

closed 2023-03-06 by DzikuVx

geofrancis, 2023-12-30:
has this still never been implemented? been trying for days to get an external barometer working.