iNavFlight/inav #5671: CLI oddities when accessed over Serial BlueTooth Bridge

PaulWieland, 2020-05-04:
## Current Behavior
My flight controller is buried in my plane, so its extremely difficult to plug in with a cable. 
To avoid having to physically plug in, I am using the bluetooth serial bridge with my crossfire transmitter (laptop <-> CrossFire Transmitter <-> CrossFire Receiver Serial Link <-> Flight Controller) 

Using iNav configurator or SpeedyBee or MobileFlight all result in the same strange behavior. If you type `diff`, `dump` or `get launch`, not every property is displayed on the screen.

Another problem is that the `gyro_hardware_lpf` setting will randomly be lost after saving other unrelated variable changes in the CLI. I have to reconnect and set the `gyro_hardware_lpf` parameter and save again at which point it's fine.

Again, the front end client doesn't seem to matter as both the iNav app on my Mac and the SpeedyBee/Mobile Flight app on my iPhone result in the same behavior.

## Steps to Reproduce

1. Connect to FC over bluetooth with iNav Configurator/SpeedyBee/MobileFlight
2. Go to CLI
3. type `dump` or `get launch`
4.

## Expected behavior
`get launch` should output a list like this:
```
# get launch
nav_fw_launch_velocity = 300
Allowed range: 100 - 10000

nav_fw_launch_accel = 1863
Allowed range: 1000 - 20000

nav_fw_launch_max_angle = 45
Allowed range: 5 - 180

nav_fw_launch_detect_time = 40
Allowed range: 10 - 1000

nav_fw_launch_thr = 1750
Allowed range: 1000 - 2000

nav_fw_launch_idle_thr = 1800
Allowed range: 1000 - 2000

nav_fw_launch_motor_delay = 100
Allowed range: 0 - 5000

nav_fw_launch_spinup_time = 50
Allowed range: 0 - 1000

nav_fw_launch_min_time = 0
Allowed range: 0 - 60000

nav_fw_launch_timeout = 10000
Allowed range: 0 - 60000

nav_fw_launch_max_altitude = 0
Allowed range: 0 - 60000

nav_fw_launch_climb_angle = 25
Allowed range: 5 - 45
```

But instead it outputs this:
```
# get launch
nav_fw_launch_velocity = 300
Allowed range: 100 - 10000

nav_fw_launch_accel = 1863
Allowed range: 1000 - 20000

nav0

nav_fw_launch_timeout = 10000
Allowed range: 0 - 60000

nav_fw_launch_max_altitude = 0
Allowed range: 0 - 60000

nav_fw_launch_climb_angle = 25
Allowed range: 5 - 45
```

## Suggested solution(s)
I'm not sure how to solve.

## Additional context

TBS CrossFire Serial BlueTooth Bridge
Tested with both Nano & Full Diversity receivers

- Matek F405 Wing & Matek F405 Mini both
- iNav Version 2.4.0

stronnag, 2020-05-04:
You should take this up with the crossfire vendor. inav and the CLI, configurator, ground stations like mwp or ezgui work perfectly OK over bluetooth and bluetooth bridges with 3DR / HC12 radio links.

PaulWieland, 2020-05-06:
> You should take this up with the crossfire vendor. inav and the CLI, configurator, ground stations like mwp or ezgui work perfectly OK over bluetooth and bluetooth bridges with 3DR / HC12 radio links.

Would it be possible for you to share a video or something of it working so I can take that to TBS to prove its crossfire that is at fault?

stronnag, 2020-05-06:
I don't see what a video would demonstrate;  the fact that I have thousands of flight logs collected by MSP / LTM over 3DR radio and almost 5 years says something about  its effectiveness as a transparent serial link.

TBS on the other hand have something of a history of failure and denial over transparent serial, like their "inject large amounts random MAVLINK into MSP streams" stunt. @wx4cb could tell you about the hoops he had to go through to get them to even acknowledge their bugt, let alone address it, let alone the amount of my time he consumed by raising it as a (not a) mwp bug.

Anywhere, here's some hard evidence:

* diff-usb.txt is a `diff` captured via USB
* diff-bt.txt is a `diff` captured via a BT-3DR radio link.

Guess what, apart from the terminal invocation at the top (which is nothing to do with inav), they are identical, all c. 3Kb, 170 lines of the diff. No corruption, no missing lines, just pure, unadulterated, correct, valid `diff` over a transparent serial wireless link. 

This is not an inav issue, the CLI works just fine. I think we can close this.

[diff-bt.txt](https://github.com/iNavFlight/inav/files/4588911/diff-bt.txt)
[diff-usb.txt](https://github.com/iNavFlight/inav/files/4588912/diff-usb.txt)

closed 2020-05-06 by stronnag

wx4cb, 2020-05-06:
as stronnag said, this is most likely a crossfire issue. however with respect. make sure you are NOT running at 115k2 baud, use 57k6 or lower.

However that being said, i can connect perfectly fine via crossfire bluetooth using the inav configurator..... through crossfire when using an HC05 programmed up to automatically connect to the crossfire.

this is how my tracker is working and works flawlessly. using MSP when not armed, and Mavlink when it is armed. 

Please be aware that if the module is in FMD1 or lower, serial bridge will not work as expected, that is a known issue and will NOT be looked at by TBS as their support is for mavlink. 

Also make sure that you are running the latest firmware from tbs

PaulWieland, 2020-05-06:
@wx4cb thank you for the tips. Appreciated.

I am using 57k baud, as its the only rate that will connect.
Per your advice I just checked the firmware and found that there is an update, so I'm going to try that next. 

Can you explain what FMD1 is?

wx4cb, 2020-05-06:
> Can you explain what FMD1 is?

FMD are the flight modes crossfire can be in, in RFMD1, serial bridge doesn't work:

RFMD = 2 , 150Hz Mode,
RFMD = 1, 50Hz Mode
If RFMD = 0, 4Hz Mode (no telemetry)

PaulWieland, 2020-05-06:
Okay gotcha. Its set to dynamic. When I update the settings with the configurator, the plane and transmitter are within feet of my computer, so I would think it would be in 150Hz mode.

Next question is about mavlink - ashamed to say I know nothing about it. My crossfire serial port is wired to a UART and that UART is set to MSP. Mavlink is not enabled.

Do I need to use mavlink in my situation? I simply want to avoid having to plug the plane in (FC is hard to reach) to adjust settings/waypoints, etc. I don't use the serial connection when flying.

jaako, 2020-05-12:
I had a similar issue.  I was looking at the iNav GPS tab, connected over CRSF bluetooth (9600), when the configurator suddenly disconnected.  Later, I went to arm and gyro_hardware_lpf was corrupted.  I've read a lot about how poor the CRSF bluetooth implementation is, but couldn't iNav be more resilient to garbage serial data and why is gyro_hardware_lpf the only setting that gets corrupted?