iNavFlight/inav #5679: Dshot does not work with more than 4 motor

mnd31, 2020-05-05:
I know dshot is still in development with iNav, but I just tried to use it (any Dshot) on my new Matek722SE hex build and could never get a signal out of output S5. I tried swapping ESCs around and the issue always followed the S5 output. The motor will not spin. BHeliSuite sees all 6 ESCs correctly.

If I switch to any other protocol works it works as expected. 

Will use multishot for now.

digitalentity, 2020-05-06:
This is a hardware limitation of 722SE board. Unfortunately S5 can't be used for DSHOT.

mnd31, 2020-05-06:
Interesting, good to know, thanks! Is that documented somewhere? Can S7 be used for DSHOT instead?

JeanFPV, 2020-05-08:
http://www.mateksys.com/?p=5159#tab-id-7
INAV Target Tips
S5 doesn’t work for Hex/Octo motor due to normal DMA confliction if Dshot is selected
Use multishot for Hex/Octo copter
BF uses burst DMA, no confliction between S1 ~S8 Dshot

For iNav: Anyway we can't use dshot (me too have hexabuild), bcs remapping in not avaliable.
For BF: We can use dshot with remap s5 to s7 pin in theory (but I did not check), but if we want use RPMfiltering - this problem will manifest itself and will not allow.
P.S. I think Hexa/Octa build is not mainstream and because that we have some troubles with build and setting... But I’m not upset, because the experience of using RPM did not give me impressive results, and multishot does its job well (do not forget to just calibrate the ESCs, manually set the PWM thresholds)

mnd31, 2020-05-09:
Thanks! I agree multishot works well enough. My hex is flying great with it.

closed 2020-05-09 by mnd31

pDR8iLQ4go, 2020-05-10:
I have a hex as well. Using a Marek F405 CTR with multishot. Does the F405 controller support dshot on S5 and S6 output?

JeanFPV, 2020-05-10:
> I have a hex as well. Using a Marek F405 CTR with multishot. Does the F405 controller support dshot on S5 and S6 output?

https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20MatekF405.md#hardware-features

- 6 PWM / DSHOT capable outputs
Looks like yes.