iNavFlight/inav #5681: Motor cuts out in NAV modes

asangag, 2020-05-05:
## Current Behavior

 **INAV 2.4.1 on Matek F405 wing on a  Plane**

I can arm and fly in manual , acro or horizon mode without any issues but as soon as I activate nav cruze, angle ,alt hold or pos hold mode the motor cuts out completely!  As soon as I exit any of those nav modes and go back to acro, manual or horizon motor starts again.  I have used INAV on Omnibus F4 pro before and never had this issue!   Any ideas ??

cheers

Below is my dump

# dump

# version
# INAV/MATEKF405SE 2.4.0 Feb 11 2020 / 22:58:10 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 100 0 -1
smix 1 4 0 100 0 -1
smix 2 5 2 100 0 -1
smix 3 2 1 100 0 -1

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 -100
servo 4 1000 2000 1500 -100
servo 5 1000 2000 1500 -100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# logic
logic 0 0 0 0 0 0 0 0
logic 1 0 0 0 0 0 0 0
logic 2 0 0 0 0 0 0 0
logic 3 0 0 0 0 0 0 0
logic 4 0 0 0 0 0 0 0
logic 5 0 0 0 0 0 0 0
logic 6 0 0 0 0 0 0 0
logic 7 0 0 0 0 0 0 0
logic 8 0 0 0 0 0 0 0
logic 9 0 0 0 0 0 0 0
logic 10 0 0 0 0 0 0 0
logic 11 0 0 0 0 0 0 0
logic 12 0 0 0 0 0 0 0
logic 13 0 0 0 0 0 0 0
logic 14 0 0 0 0 0 0 0
logic 15 0 0 0 0 0 0 0
[... 1154 lines cut ...]
set roll_rate = 18
set pitch_rate = 20
set yaw_rate = 9
set manual_rc_expo = 50
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

# end the command batch
batch end

giacomo892, 2020-05-06:
Blackbox log please, and next time use social media for support requests

asangag, 2020-05-07:
Ok thanks. I have the blackbox logs but cant attach them as they are too big files . How do i do it ?
thanks

asangag, 2020-05-08:
here is the link to the log file. its too big to upload here

https://drive.google.com/file/d/1w2T_Utc7gFo6sYcEZwOc3JzcJS0E4Uv9/view?usp=sharing

AnandNL, 2020-05-09:
> set nav_fw_cruise_speed = 0 

thats too low

breadoven, 2020-05-13:
Looking at the log file your received RC throttle signal (rcData[3]) drops below 1000 in the modes that cause the motor to stop so iNAV simply thinks the throttle stick is at Min Throttle and cuts the motor. Angle isn't a NAV mode so don't think this is NAV related more likely some setup issue with your transmitter and mode switching.

asangag, 2020-05-21:
> rcData[3]

I had the throttle stick around 50% when I flicked the switch to change the mode to nav_cruze  , angle or nav RTH. So something else is forcing the throttle signal to go below 1000. ?  

I re-flashed the board with older Inav 2.3CR and tested it. This time nav RTH , failsafe modes worked and automatic throttle kicked in to bring the craft home but nav cruze , nav position hold and angle modes all still cutting the throttle to zero.  This is very strange !

breadoven, 2020-05-21:
RTH auto throttle should work even with the throttle at minimum. You're saying it didn't with 2.4.1 ?

Can you plug something (servo) into the Rx directly to see what it does. rcData(3) is supposed to be the raw signal from the Rx so unless iNav is doing something odd with the SBUS signal it would seem to be an RC problem. You should also see the problem in the Rx tab of Configurator, what does it show ?

asangag, 2020-05-21:
> RTH auto throttle should work even with the throttle at minimum. You're saying it didn't with 2.4.1 ?
>
> Can you plug something (servo) into the Rx directly to see what it does ? rcData(3) is supposed [...]

Ok I found the problem!! thanks to your lead on the throttle signal. I checked my Tx (jumper T16) and noticed every time i flicked the switch assigned to angle mode it cuts the throttle out put on the Tx. so i further checked the mixer page , special functions page and logical switches page. i found a residue logical switch from my old throttle safety function still active. 

When I was setting up Inav, I was too lazy to create a new model and copied an old model and modified it . I deleted my throttle safety function but missed one logical switch which coincidentally got assigned to the angle mode switch. So thats why the Tx cut the throttle every time i flicked the switch for angle mode . I also had a logical switch configuration to voice out airmode when none of the switches are active, which again coincidentally  involved that residue throttle cut logical switch. so all switches assigned to cruise, atl hold, posi hold had the throttle cut logical switch embedded in them. 

I feel like an absolute idiot to waste everyone's time trying to resolve this issue.!!  Thanks everyone  for all your help ! 

Lesson learned ...not to cut corners!  lol  I deleted the complex logical switch system and now have a simple setup and everything is working as it should be! :) 

As to why the RTH or Fail safe did not work with Inav  2.4.1 but works with 2.3 is probably because I fiddled around with nav_fw_min_thr and nav_fw_max_thr and might have caused some errors ? Thats all I can think of.  Now that I have everything working I might re-flash latest Inav and see how it goes.

Thanks again for all your support!

closed 2020-06-11 by DzikuVx