iNavFlight/inav #5686: "Unable to Arm / Sensors Calibrating" needs to be more verbose after a time

FireNWater, 2020-05-07:
## Current Behavior

## Desired Behavior

## Suggested Solution

## Who does this impact? Who is this for?

## Additional context

FireNWater, 2020-05-07:
When the FC cannot arm due to a sensor problem, the OSD message "Unable to Arm / Sensors Calibrating" shows on the OSD.  Since the flight vehicle could have many sensors, I would like the message to show which sensor is currently keeping the system from arming.  This would greatly enhance troubleshooting, especially in the field.

Pairan, 2020-05-07:
Had the same in the field lately ... "RIGHT PITCH NOT CENTERED" ...

I had connected a 4s instead of a 3s battery ... changing that solved my problem somehow,

closed 2023-03-06 by DzikuVx

Tgoraj, 2024-02-10:
Ok, so plugging a 3s battery in solved it but my plane only uses 4s batteries so… what do I do?why is there no solution to this issue?

MrD-RC, 2024-02-10:
Hold the aircraft still. Its calibrating the onboard sensors. It can’t calibrate the gyro if it’s moving.