iNavFlight/inav #5693: Acc/Gyro Aligment issue Matek F411 Mini SE

btandrei, 2020-05-08:
## Current Behavior

Default or CW180 sensor alignment setting has the same response, board does indeed respect marked orientation but if the board is rotated 180 it can't be calibrated and compass doesn't work correctly.     

## Steps to Reproduce

1.Set align_acc/gyro=CW180 has no effect, board keeps default sensor orientation
2.Set align_acc/gyro=CW0, sensor alignment now responds with CW180 rotation

## Expected behavior

## Suggested solution(s)

Check this settings and make sure align_acc/gyro has correct response.

## Additional context

Also the latest Inav Configurator 2.4 doesn't have sensor alignment tab anymore like 2.3 had.
---

- FC Board name and vendor: Matek F411 Mini SE
- INAV version string: MATEKF411SE
- INAV/MATEKF411SE 2.4.0 Feb 11 2020 / 23:07:15 (bcd4caef9)
- GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

digitalentity, 2020-05-08:
Default sensor orientation for MATEKF411SE is `CW180_DEG`, @MATEKSYS can you please check if the correct one is CW0_DEG?

btandrei, 2020-05-08:
If it's set to CW180 or DEFAULT the orientation then matched the arrow on the PCB. By default it says align_acc = DEFAULT and you don't know that it's already rotated, and setting CW180 did nothing and the calibrations didn't work because i got the board rotated with USB port on the other side.

digitalentity, 2020-05-08:
This is WAI. `DEFAULT` never implies `CW0_DEG`. 
`gyro_align`, `acc_align` etc are to define orientation of **the sensor chip relative to the board** and not the board rotation. Use `align_board_***` to change the orientation of the whole board relative to the aircraft.

closed 2020-05-08 by digitalentity

btandrei, 2020-05-08:
I've tried that already to set align_board_yaw = 1800, acc/gyro orientation seems ok, but then compass will have an error of 180degrees N points to S and the Heading is not stable when tilting the quad. The only fix was to work with acc/gyro_align. Inav Configurator 2.3 did have these options in the GUI to set board orientation and worked.

digitalentity, 2020-05-08:
Since the compass is a different piece of hardware, if you rotate the board, alignment of the compass sensor **relative to the board** changes. You either need to rotate the board and the compass together or tweak `mag_align`,.

Please consider reading the [wiki](https://github.com/iNavFlight/inav/wiki/GPS--and-Compass-setup) and using social channels (Telegram channel, RC Groups etc) for support and configuration issues. This is a bug tracker and what you are experiencing is neither a bug nor a feature request.

MATEKSYS, 2020-05-08:
@digitalentity  Default gyro alignment is right for F411-WSE and F411-miniSE when arrow is facing forward.