iNavFlight/inav #5695: CC3D Revolution: Do not configure the Servo output

ricardovalmeida, 2020-05-08:
Hi friends
I am in trouble with my cc3d revolution.
I can't use the mix to trigger another servo output, when I add another output, the tricopter doesn't set anymore and I have no signal on the outputs, S5 and S6, tricopter configuration, I'm using the INAV / REVO version 2.4.0 Feb 11 2020
you can help?
Thank you
[image]

DzikuVx, 2020-07-07:
Checked with oscilloscope and servos were assigned correctly

closed 2020-07-07 by DzikuVx

ricardovalmeida, 2020-07-07:
Hi friend, I already checked there is no pwm signal for the servo motor, you can help me, configure the inav or it may be the firmware that is not working.
thank you.

Ricardo Almeida

Enviada em: terça-feira, 7 de julho de 2020 14:13
Para: iNavFlight/inav
Cc: Ricardo; Author

 

Este e-mail foi verificado quanto a vírus pelo AVG.
http://www.avg.com