iNavFlight/inav #5708: DSHOT unavailable for Omnibus F7 V2

DBetic-Pilot, 2020-05-10:
## Current Behavior
Set up a HGLC Wind5 quad as a 5" GPS drone using the 2-12-2020 firmware and INAV 2.5.0 RCI configurator. When setting up motors/outputs, if i select any DSHOT protocols (the board should handle up to 1200, but none work), it will say that it has saved to EEPROM. After a reboot, it has defaulted back to "STANDARD."

## Steps to Reproduce

1.Set ESC protocol to any DSHOT protocol
2.click save or save and reboot
3.observe

## Expected behavior
I would expect that the board should be able to handle it, as it was running DSHOT out of the box with Betaflight. 

## Suggested solution(s)
- Not sure if this is a problem with the firmware file, or some miscommunication/setting with INAV not recognizing the OMNIBUS F7V2 as supporting those protocols, as it is just now supported in 2.5. I upgraded configurator hope this would help but it did not. 

## Additional context
Dump at https://pastebin.com/PnbJgZWz

---

# INAV/OMNIBUSF7V2 2.4.0 Feb 11 2020 / 22:36:01 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]
Thanks in advance
Chris

digitalentity, 2020-05-11:
Fixed in 2.5 already

closed 2020-05-11 by digitalentity

DBetic-Pilot, 2020-05-11:
@digitalentity 
When you say this is fixed in 2.5, which version of 2.5 are you referencing? I was running INAV 2.5_RC when the errors showed up. is there another release i should be testing against?
Thanks

digitalentity, 2020-05-11:
You were running version 2.4.0, not 2.5.0-RC1

```
# version
# INAV/OMNIBUSF7V2 2.4.0 Feb 11 2020 / 22:36:01 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]
```