iNavFlight/inav #5710: OMNIBUS F4 FIREWORKS V2 IMU

Ex4nder, 2020-05-10:
## Current Behavior
The Imu don't work

## Steps to Reproduce
Just flash the firmware

## Expected behavior
MPU6500 should be detected and work, but it's not.

## Suggested solution(s)
I tried to check the target.h, but the resources are correct, but still no go.

## Additional context

in betaflight it works perfectly

AIRBOT OMNIBUS F4 FIREWORKS V2

# version
# INAV/FIREWORKSV2 2.5.0 May  9 2020 / 14:48:35 (272de567d)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

Entering CLI Mode, type 'exit' to return, or 'help'

# version
# INAV/FIREWORKSV2 2.5.0 May  9 2020 / 14:48:35 (272de567d)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# dump all

# version
# INAV/FIREWORKSV2 2.5.0 May  9 2020 / 14:48:35 (272de567d)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# resources

# mixer

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# logic
logic 0 0 -1 0 0 0 0 0 0
logic 1 0 -1 0 0 0 0 0 0
logic 2 0 -1 0 0 0 0 0 0
logic 3 0 -1 0 0 0 0 0 0
logic 4 0 -1 0 0 0 0 0 0
[... 1405 lines cut ...]
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

# battery_profile
battery_profile 3

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

# restore original profile selection
profile 1
battery_profile 1

# save configuration
save

#

digitalentity, 2020-05-11:
So we understand correctly: the board doesn't detect gyro, correct?

Ex4nder, 2020-05-11:
exactly, it's not detected, red on the first boot, and then hardware goes to "none"

digitalentity, 2020-05-11:
Can you also post an output of "status" command?

Ex4nder, 2020-05-11:
Hi, not now, but no errors on the status, since after the first boot it goes to NONE.
Should I set Gyro hardware to MPU6500 and reboot and then print the status?

digitalentity, 2020-05-11:
Do a flash with full erase, go to CLI, execute "status" and post the output here. Thanks!

Ex4nder, 2020-05-11:
I guess after reconfiguring back the gyro as MPU6500, so first boot I will chose "mini quad", reset, choose gyro again, and then execute status

digitalentity, 2020-05-11:
Please don't modify any settings manually.

I don;t have that board here and I'm trying to figure out what's going on here and for that I need to understand exactly at what step it breaks.

Ex4nder, 2020-05-11:
If you want, this evening after work we can make a teamwiever session, so you can see live, I've a linux machine, not so fast, to recompile and try.

digitalentity, 2020-05-11:
I think it would be better if we continue on Github so other developers could contribute to troubleshooting this as well.

So, the steps to do to continue:

1. Flash INAV 2.5 with full erase
2. On first connection Configurator will ask for preset. Chose "Keep current settings"
3. Go to CLI and execute "status"

Thanks.

Ex4nder, 2020-05-11:
perfect, this evening I will report

Ex4nder, 2020-05-11:
Status of betaflight 
# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.29V, Core temp=25degC
Stack size: 2048, Stack address: 0x1000fff0
Config size: 3960, Max available config: 16384
Gyros detected: gyro 1
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280, MAG=QMC5883
OSD: MAX7456
System Uptime: 125 seconds, Current Time: 2020-05-11T17:28:37.352+00:00
CPU:12%, cycle time: 125, GYRO rate: 8000, RX rate: 33, System rate: 10
Voltage: 44 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: RXLOSS ANGLE CLI MSP

Status of Inav  (but after choosing keep current settings it rebooted)
status
System Uptime: 24 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.31V (1S battery - NOT PRESENT)
CPU Clock=168MHz, BARO=BMP280, MAG=QMC5883
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=UNAVAILABLE, ACC=NONE, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
Stack size: 6144, Stack address: 0x10010000, Heap available: 1948
I2C Errors: 0, config size: 5803, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 1, used = ADC 1
  AIRSPEED : configured = none, used = none
System load: 8, cycle time: 102, PID rate: 9803, RX rate: 49, System rate: 9
Arming disabled flags: ANGLE CAL COMPASS HWFAIL RX CLI
VTX: not detected

# 
After selecting mpu6000 and rebooting again:

Entering CLI Mode, type 'exit' to return, or 'help'

# status
System Uptime: 24 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.31V (1S battery - NOT PRESENT)
CPU Clock=168MHz, BARO=BMP280, MAG=QMC5883
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=UNAVAILABLE, ACC=UNAVAILABLE, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
Stack size: 6144, Stack address: 0x10010000, Heap available: 1948
I2C Errors: 0, config size: 5803, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 1, used = ADC 1
  AIRSPEED : configured = none, used = none
System load: 9, cycle time: 101, PID rate: 9900, RX rate: 49, System rate: 9
Arming disabled flags: ANGLE CAL COMPASS HWFAIL RX CLI
VTX: not detected

# 
after select mpu6500 and rebooting again...

Entering CLI Mode, type 'exit' to return, or 'help'

# status
System Uptime: 20 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.31V (1S battery - NOT PRESENT)
CPU Clock=168MHz, BARO=BMP280, MAG=QMC5883
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=UNAVAILABLE, ACC=UNAVAILABLE, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
Stack size: 6144, Stack address: 0x10010000, Heap available: 1948
I2C Errors: 0, config size: 5803, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 1, used = ADC 1
  AIRSPEED : configured = none, used = none
System load: 8, cycle time: 100, PID rate: 10000, RX rate: 49, System rate: 9
Arming disabled flags: ANGLE CAL COMPASS HWFAIL RX CLI
VTX: not detected

# 

and for your refererence

# resource
IO:
----------------------
A00: FREE 
A01: FREE 
A02: FREE 
A03: FREE 
A04: SPI_PREINIT CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: LED1 OUT
A09: SOFTSERIAL1 UART TX/RX
A10: SERIAL1 UART RX
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: OSD CS
B00: FREE 
B01: FREE 
B02: FREE 
B03: BARO CS
B04: BEEPER OUT
B05: FREE 
B06: FREE 
B07: FREE 
B08: FREE 
B09: FREE 
B10: I2C2 SCL
B11: I2C2 SDA
B12: FLASH CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: FREE 
C01: ADC CH1
C02: ADC CH2
C03: FREE 
C04: FREE 
C05: FREE 
C06: FREE 
C07: FREE 
C08: FREE 
C09: FREE 
C10: SPI3 SCK
C11: SPI3 MISO
C12: SPI3 MOSI
C13: FREE 
C14: FREE 
C15: FREE 
D00: FREE 
D01: FREE 
D02: SPI_PREINIT CS
D03: FREE 
D04: FREE 
D05: FREE 
D06: FREE 
D07: FREE 
D08: FREE 
D09: FREE 
D10: FREE 
D11: FREE 
D12: FREE 
D13: FREE 
D14: FREE 
D15: FREE

digitalentity, 2020-05-11:
I've found a FIREWORKSV2 board in my FC box and interestingly, after erasing and flashing, it works:

```
# version
# INAV/FIREWORKSV2 2.5.0 May 11 2020 / 20:58:34 (2a9058b5f)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# status
System Uptime: 58 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.34V (1S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
Stack size: 6144, Stack address: 0x10010000, Heap available: 1928
I2C Errors: 0, config size: 5803, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 1, used = ADC 1
  AIRSPEED : configured = none, used = none
System load: 2, cycle time: 1002, PID rate: 998, RX rate: 49, System rate: 9
Arming disabled flags: ACC RX CLI
VTX: not detected
```

However, it detects MPU6500 gyro, not MPU6000 as on your board. It's possible that your board has a different IMU chip, in which case it's "correctly" not detecting it as it expects MPU6500 on that SPI bus device slot.

Seems easily fixable

Ex4nder, 2020-05-11:
What do you suggest? Add use imu mpu6000 in the target.h?

digitalentity, 2020-05-11:
Try this build: [inav_2.5.0_FIREWORKSV2.zip](https://github.com/iNavFlight/inav/files/4611714/inav_2.5.0_FIREWORKSV2.zip)

It enabled detection of both MPU6000 and MPU6500 on both SPI bus device slots. Note that you may have to try changing `gyro_to_use` setting in CLI to check the second slot.

Ex4nder, 2020-05-11:
[image]

# status
System Uptime: 34 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.31V (1S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280, MAG=QMC5883
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
Stack size: 6144, Stack address: 0x10010000, Heap available: 1928
I2C Errors: 0, config size: 5803, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 1, used = ADC 1
  AIRSPEED : configured = none, used = none
System load: 2, cycle time: 501, PID rate: 1996, RX rate: 49, System rate: 9
Arming disabled flags: ANGLE COMPASS ACC RX CLI
VTX: not detected

TOP!, can you post the difference to the target file to understand how to do? thank you!

digitalentity, 2020-05-12:
Awesome, the change #5717 is ready to be merged, you can have a look at that PR to understand what changed.

closed 2020-05-12 by digitalentity