iNavFlight/inav #5714: Inav 2.5 and omnibusf4v3 yaw spin on takeoff.

NasirHayat, 2020-05-11:
I am super excited to try this new version. Today I did install 2.5 at omnibusf4v3 5inch quad. which was working fine at 2.4. I did a new configuration noting took from the old version all stock configuration pid etc. When it takeoff and start yaw spin to left and spins really fast. Now shifted back to 2.4 and everything becomes fine. Please I will really appreciate it if some guide me on how to fix this issue.

digitalentity, 2020-05-11:
Please post dumps for 2.4 (working) and 2.5 (not working) so we can compare.

NasirHayat, 2020-05-11:
> Please post dumps for 2.4 (working) and 2.5 (not working) so we can compare.

Thanks for your suggestion but the developer who is part of the development team suggests not copying anything from the previous version as this new version has some massive changes in code.

digitalentity, 2020-05-11:
We can not debug this without looking at your **previous** configuration. There could be hardware-related settings that were accounted for in your working configuration (flyable on 2.4) vs current.

DzikuVx, 2020-05-11:
Please record a blackbox log and attach a config dump

DzikuVx, 2020-05-11:
@NasirHayat see if this is not the same as #5715

NasirHayat, 2020-05-11:
> @NasirHayat see if this is not the same as #5715

Thanks for your great support going to try what you suggest. As per my understanding is do full erase (I already did full erase) and not to use preset.

closed 2020-05-11 by DzikuVx