iNavFlight/inav #5716: Holybro Kakute F7 mini - CPU overflow at 4k or 8k Looptime

Glaubfox, 2020-05-11:
After flashing INAV 2.5 RC1 with full erase and factory reset on Holybro Kakute F7 Mini version 1 with K7 ;

 I get a CPU overflow if i set the Looptime to 4k or 8k. 
Tasks in Cli:  1 -     GYRO/PID    2898     361     344  105.1%  100.1%     10915

Tryed with Board only and in Full stack. 
Works with 2K Looptime (CPU at 4% load).

version
# INAV/KAKUTEF7MINI 2.5.0 May  9 2020 / 14:38:21 (272de567d)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

Thx 
for all your work for us ;-)

DzikuVx, 2020-05-14:
@Glaubfox `dump` please

Glaubfox, 2020-05-14:
3inch geprc Frame
T-Motor F15 6000kv
Ori32 4in1 ESC 
Crossfire Nano RX
Runcam Split
Matek GPS+Compass
AKK Ultimate mini VTX

all together long range mini Quad ;-) 

No Problems with Betaflight in 4k or 8k Looptime

DzikuVx, 2020-05-14:
@Glaubfox the dump you provided looks like 100% defaults without MAG or GPS you write about configured. 
Still, I flahed my F722MINI and applied the dump you provided: load 4% at 4kHz.

If you have MAG enabled, it has to be powered and I2C speed should be raised to 800kHz. Still, for a good flight performance 2kHz looptime is enough

closed 2020-05-27 by DzikuVx