iNavFlight/inav #5719: have auto trim ignore "servo" outputs that are set to direct RC channel

wx4cb, 2020-05-12:
## Current Behavior
Current behavior is that auto trim will also "trim" the midpoint for channels that are set to "RC channel xx" in the mixer tab to whatever value they're at when auto trim is set.

For example, my Servo 5 (lights channel) is set to RC Channel 10 if it's "off (-100/1000uS) then that's what the mid point gets set to. if it's "on" (100/2000uS) then that's what the midpoint gets set to.

[image]

## Desired Behavior
auto trim to ignore any servo mixer set to a specific RC channel

## Suggested Solution
ignore any servo output that is specifically set to an RC channel

## Who does this impact? Who is this for?
Anyone that uses a pan/tilt, PWM controlled switches or lights etc

wx4cb, 2020-07-12:
Bump. Is this possible?

avsaase, 2021-03-13:
I'm working on a new auto trim feature that will also make it configurable which servos are trimmed.

avsaase, 2021-05-08:
Closed by https://github.com/iNavFlight/inav/pull/6708.

closed 2021-05-08 by avsaase